OpenEtherCATsociety/SOEM #522: Is it possible to run SOEM master code on a TI micro-controller?

dyan49, 2021-06-03:
Hey guys, I'm planning to program SOEM master code on a TI micro-controller, probably in their C6xxx-series. Does anyone of you have experience in that? I only saw successful implementation of SOEM code on CPU under Linux or Microsoft OS. If you guys ever worked on it, could you share your experience of how to do it? Thank you very much for the help!

ArthurKetels, 2021-06-03:
SOEM has been implemented on a variety of micro-controllers.  As you can see in the OSHW abstraction folder there are already multiple RT-OS supported that run on micro-controller. Even bare-bone implementation is fairly quick to do. Minimum requirements 8-bits, 32kB ram, 128kB flash, 100mb/s ethernet (direct, spi or via usb). Look in OSHW and OSAL folders to see what needs to be implemented.

dyan49, 2021-06-03:
Hi Arthur, thanks for the reply. I've just looked in the folder you mentioned "OSHW" and "OSAL". However, I didn't find any abstraction folder that we can directly use for TI micro-controller TMS320C6727 [](https://www.ti.com/product/TMS320C6727?utm_source=google&utm_medium=cpc&utm_campaign=epd-null-null-GPN_EN-cpc-pf-google-wwe&utm_content=TMS320C6727&ds_k=%7b_dssearchterm%7d&DCM=yes&gclid=Cj0KCQjw--GFBhDeARIsACH_kdY_oGSEJiq7FZtwa7o3VrkKUNabG0nhX4ClmharKHZv2ID1PDX-mUcaAtNEEALw_wcB&gclsrc=aw.ds). Since you are the developer of the SOEM, do you know whether the TI-RTOS which is specifically for TI controller is integrated in the folder? Thank you for the clarification!

phenom8305, 2021-06-04:
Dong-hyun Lee님이 Gmail 비밀 모드를 통해 이메일을 보냈습니다.

[image: Gmail 로고]Re: [OpenEtherCATsociety/SOEM] Is it possible to run SOEM master code on a TI micro-controller? (#522) <https://confidential-mail.google.com/msg/AG7J_YRgp5nIyLqhUpTsNAmghTOBRTOcpDqzZyQoG5FblrGnoWjViaovVBGKCcuBvMDCMfVLRb_LzJsm__mNGLoVrBQl2FEwSxohbckyr3ykmiU0mLh1b1McpRxc7zjxlQJm9E8o2kPVQN9SMfFF5V2KmLyoG-JKpbAzTmmTCj3hxpgT68bntnY42jti2JB9xuVswS8WU2llI34=>

이 메일은 2021. 6. 3., 오후 5시 6분 16초 GMT-7에 전송되었습니다.
아래 링크를 클릭하면 메일을 열 수 있습니다. 이 링크는 ***@***.***에서만 열 수 있습니다.

이메일 보기 <https://confidential-mail.google.com/msg/AG7J_YRgp5nIyLqhUpTsNAmghTOBRTOcpDqzZyQoG5FblrGnoWjViaovVBGKCcuBvMDCMfVLRb_LzJsm__mNGLoVrBQl2FEwSxohbckyr3ykmiU0mLh1b1McpRxc7zjxlQJm9E8o2kPVQN9SMfFF5V2KmLyoG-JKpbAzTmmTCj3hxpgT68bntnY42jti2JB9xuVswS8WU2llI34=>

Gmail 비밀 모드를 사용하면 보내는 메일을 더 세부적으로 관리할 수 있습니다. 이 메일을 보낸 사람이 메일에 만료 시간을 설정했거나, 메일을 인쇄 또는 전달하지 못하도록 설정했거나 메일에 대한 액세스를 추적 중일 수 있습니다. 자세히 알아보기 <https://support.google.com/mail/answer/7674059>
Gmail: Google이 지원하는 이메일
사용 시 Google 개인정보처리방침 <https://myaccount.google.com/privacypolicy?hl=ko>이 적용됩니다.
Google LLC, 1600 Amphitheatre Parkway, Mountain View, CA 94043, USA
이 메일은 누군가가 Gmail 비밀 모드를 통해 나에게 이메일을 보냈기 때문에 전송되었습니다.
[image: Google 로고]

phenom8305, 2021-06-04:
Your Google verification code is 898545

ArthurKetels, 2021-06-04:
There are many targets that have been developed for SOEM and are not in the public repository. Only when people take the effort to make a pull-request they will be included. There are thousands of SOEM users we never hear of.

You have the choice to make your own abstraction layer for your target or ask a professional party to do that for you. RT-labs is such a partner that can support you with this.

nakarlsson, 2021-07-21:
@dyan49 , can this issue be closed?

nakarlsson, 2022-01-14:
close due to inactivity

closed 2022-01-14 by nakarlsson