OpenEtherCATsociety/SOEM #524: Segmentation fault after changing state to operational.

TCWilliamson, 2021-06-12:
I'm just getting started with SOEM, using Ho Tam's SOEM with a Raspberry Pi.

So far I've got it detecting an ethercat device, and changing states (to an extent) however, I can't get past having 'segmentation fault' errors. I'm a bit of a noob on this kind of thing and wouldn't mind having a bit of a clarification as to what that exactly means, and what I might be missing out!

The code I'm running is a modified version of simple_test.

```

root@raspberrypi:/home/pi/soem-w5500-rpi/test/MAXPOS_CSP# ./MAXPOS_CSP
SOEM (Simple Open EtherCAT Master)
Simple test
gpio: init OK
******************************************************
Wiznet driver for RPI SOEM - V1.0
Author: Ho Tam  - thanhtam.h[at]gmail.com
******************************************************
Starting simple test
Link Check...OK
ec_init on (null) succeeded.
1 slaves found and configured.
Slave 1 has CA? false
Slave: 1 - Write at 0x6060:0 => wkc: 1; data: 0xa       {OpMode}
Slave: 1 - Read at 0x6061:0 => wkc: 1; data: 0x0 (0)    [OpMode display]
Slave: 1 - Read at 0x1c12:0 => wkc: 1; data: 0x1 (1)    [rxPDO:0]
Slave: 1 - Read at 0x1c13:0 => wkc: 1; data: 0x1 (1)    [txPDO:0]
Slave: 1 - Read at 0x1c12:1 => wkc: 1; data: 0x1600 (5632)      [rxPDO:1]
Slave: 1 - Read at 0x1c13:1 => wkc: 1; data: 0x1a00 (6656)      [txPDO:1]
Slave: 1 - Read at 0x1c12:0 => wkc: 1; data: 0x1 (1)    [rxPDO:0]
Slave: 1 - Read at 0x1c13:0 => wkc: 1; data: 0x1 (1)    [txPDO:0]
Slave: 1 - Read at 0x1c12:1 => wkc: 1; data: 0x1600 (5632)      [rxPDO:1]
Slave: 1 - Read at 0x1c13:1 => wkc: 1; data: 0x1a00 (6656)      [txPDO:1]

Slave:1
 Name:Kinco FD
 Output size: 48bits
 Input size: 48bits
 State: 2
 Delay: 0[ns]
 Has DC: 1
Slave: 1 - Read at 0x10f1:2 => wkc: 0; data: 0x0000 (0) [Heartbeat?]
Slave: 1 - Write at 0x10f1:2 => wkc: 1; data: 0x0001    {Heartbeat}
Slave: 1 - Write at 0x60c2:1 => wkc: 1; data: 0x2       {Time period}
Slave: 1 - Write at 0x2f75:0 => wkc: 1; data: 0x02      {Interpolation timeout}
Slaves mapped, state to SAFE_OP.
segments : 1 : 12 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Slave: 1 - Read at 0x6083:0 => wkc: 1; data: 0x4000 (16384)     [Profile acceleration]
Slave: 1 - Read at 0x6084:0 => wkc: 1; data: 0x4000 (16384)     [Profile deceleration]
Slave: 1 - Read at 0x6085:0 => wkc: 1; data: 0x18666 (99942)    [Quick stop deceleration]
Operational state reached for all slaves.
Slave: 1 - Read at 0x6041:0 => wkc: 1; data: 0x4270 (17008)     [*status word*]
Slave: 1 - Write at 0x6040:0 => wkc: 1; data: 0x00      {*control word*}
Slave: 1 - Read at 0x6041:0 => wkc: 1; data: 0x4270 (17008)     [*status word*]
Slave: 1 - Write at 0x6040:0 => wkc: 1; data: 0x06      {*control word*}
Slave: 1 - Read at 0x6041:0 => wkc: 1; data: 0x4231 (16945)     [*status word*]
Slave: 1 - Write at 0x6040:0 => wkc: 1; data: 0x07      {*control word*}
Slave: 1 - Read at 0x6041:0 => wkc: 1; data: 0x4233 (16947)     [*status word*]
Slave: 1 - Write at 0x6040:0 => wkc: 1; data: 0x0f      {*control word*}
Segmentation fault
root@raspberrypi:/home/pi/soem-w5500-rpi/test/MAXPOS_CSP#

```

The segmentation fault seems to occor when it tries to read the 0x6041 register to aquire the status word.

``` c++
                    WRITE(i, 0x6040, 0, buf16, 0, "*control word*"); usleep(100000);
                    READ(i, 0x6041, 0, buf16, "*status word*");

                    WRITE(i, 0x6040, 0, buf16, 6, "*control word*"); usleep(100000);
                    READ(i, 0x6041, 0, buf16, "*status word*");

                    WRITE(i, 0x6040, 0, buf16, 7, "*control word*"); usleep(100000);
                    READ(i, 0x6041, 0, buf16, "*status word*");

                    WRITE(i, 0x6040, 0, buf16, 0x0F, "*control word*"); usleep(100000);

                    READ(i, 0x6041, 0, buf16, "*status word*");
					
                    CHECKERROR(i);
                    READ(i, 0x1a0b, 0, buf8, "OpMode Display");
					
                    READ(i, 0x1001, 0, buf8, "Error");
```

ArthurKetels, 2021-06-13:
So what are you actually asking? Us reading a crystal bowl? The code you are using is a modified version of SOEM from a different repository.  The test code is for a Maxon servo drive, you have a Kinko FD (whatever that is). A segfault mostly is caused by a reference to a pointer that is no longer valid, or was never valid in the first place.  It is up to you to figure out exactly what instruction causes the segfault. Dig deeper.

closed 2021-06-13 by ArthurKetels

TCWilliamson, 2021-06-13:
That's just the name of the file - I've been rewriting the internals, running a simpletest that was modified. It's still SOEM framed, so I was hoping someone here would be kind enough to help - information on SOEM is a bit sparse. Any advice on recommendations on what to read to get a better understanding would be appreciated - there's not many people I know to go to, to ask questions...

I'm not exactly sure what a segmentation fault actually is!

ArthurKetels, 2021-06-13:
https://en.wikipedia.org/wiki/Segmentation_fault
I know, typing "segmentation fault" in google is difficult....
It is such a generic error that this not the right place to ask those questions. As far as I know there have not been any segfault related bugs in SOEM for years. So it is most likely in other code than ours.

TCWilliamson, 2021-06-13:
No need to be rude - I did search Segmentation fault several times, but I tied it in with EtherCAT, assuming it was more relation to EtherCAT Segments. Do you have any recommendations as to where to go for more information regarding getting familier with setting up SOEM - any good reads? All I'm looking for is a bit of help and a point in the right direction. Just trying to figure things out.