OpenEtherCATsociety/SOEM #525: Pick up mailbox from slaves when available

Salem07, 2021-06-22:
First of all Thank you for this really nice library. It is really a very exciting project

I have some questions und I would be very grateful if you can help with it. I have a slave that provides mailboxes from time to time. I like that my Ethercatmaster fetches mailboxes from slaves as soon as they are available. So I guess I am going to do this during the **non**-real-time process loop. The Answer in #405 suggests a separete thread for the SDO read/write and describes an approach for the treatment of data. 

My question ist: how do i know that a maillbox from slave is available for pick up. The following function of soem seems to do the job:  `int ec_mbxempty(uint16 slave, int timeout);`
Am I right? can I use this function in the real-time-process loop ? should I use it after getting the pdo from the slave (My assumption is that the indication that a mailbox is available is also transmitted by the pdo) ? 

Thanks in advance for the answer

ArthurKetels, 2021-06-22:
Better to use:
`int ecx_mbxreceive(ecx_contextt *context, uint16 slave, ec_mbxbuft *mbx, int timeout)`
Set the timeout to zero if you want the call to be non-blocking. If there is a mailbox available in the slave you get the content in *mbx, otherwise the return value is <=0. A return value of 0 indicates a mailbox that is already handled (errormbx or EoE).

You can use this function in the real-time loop with timeout = 0, otherwise it would be best to put this in a separate thread. As always, adding stuff to do will reduce your available bandwidth.

The optimal solution is to program an extra FMMU on the slave to map the mailbox out status register in PDO memory. Then you know in each PDO cycle if the mailbox is ready to read. This way you waste the least amount of bandwidth.

Salem07, 2021-06-24:
Thank you for the Answer. That has helped me a lot.

nakarlsson, 2021-07-21:
@Salem07 , can this issue be closed?

Salem07, 2021-07-21:
Unfortunately I did not have the chance to test my application but maybe in this context i have one question. 

You write "A return value of 0 indicates a mailbox that is already handled (errormbx or EoE)". what does this exactly mean for me as an EthercatMaster ? When an errormbx or EoE is present, how to access to the error message ?

Thank you in advance

nakarlsson, 2021-07-22:
The errormbox can be retrieved via the error API
https://github.com/OpenEtherCATsociety/SOEM/blob/219670218b78acf4e4a96da70d36eeb9fc7f96ca/test/linux/slaveinfo/slaveinfo.c#L482

The asynchronous EoE data should be picked up via a hook
https://github.com/OpenEtherCATsociety/SOEM/blob/219670218b78acf4e4a96da70d36eeb9fc7f96ca/test/linux/eoe_test/eoe_test.c#L54

closed 2021-07-23 by Salem07