OpenEtherCATsociety/SOEM #526: Slave doesnt initialize after several restarts

vargajanko, 2021-06-25:
Hi everyone,

I have 2 Synapticon Somanet slaves connected to SOEM. I can run my application for a few hours without problems. But after restarting my application several times, around 1 to 20 times, one slaves doesnt get initialized. After powercycle the slave works fine again. 

**This is the output of slaveinfo -map with working slaves:**

`SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eno1 succeeded.
slaves found and configured.
Calculated workcounter 6

Slave:1
 Name:SOMANET
 Output size: 280bits
 Input size: 440bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 000022d2 ID: 00000201 Rev: 0a000002
 SM0 A:1000 L:1024 F:00010026 Type:1
 SM1 A:1400 L:1024 F:00010022 Type:2
 SM2 A:1800 L:  35 F:00010064 Type:3
 SM3 A:1c00 L:  55 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  35 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000046 Ll:  55 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 1024 rd: 1024 MBX protocols : 0c
 CoE details: 0f FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6040:0x00 0x10 UNSIGNED16   Controlword
  [0x0002.0] 0x6060:0x00 0x08 INTEGER8     Modes of operation
  [0x0003.0] 0x6071:0x00 0x10 INTEGER16    Target Torque
  [0x0005.0] 0x607A:0x00 0x20 INTEGER32    Target position
  [0x0009.0] 0x60FF:0x00 0x20 INTEGER32    Target velocity
  [0x000D.0] 0x60B2:0x00 0x10 INTEGER16    Torque offset
  [0x000F.0] 0x2701:0x00 0x20 UNSIGNED32   Tuning command
  [0x0013.0] 0x60FE:0x01 0x20 UNSIGNED32   Physical outputs
  [0x0017.0] 0x60FE:0x02 0x20 UNSIGNED32   Bit mask
  [0x001B.0] 0x2703:0x00 0x20 UNSIGNED32   User MOSI
  [0x001F.0] 0x60B1:0x00 0x20 INTEGER32    Velocity offset
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0046.0] 0x6041:0x00 0x10 UNSIGNED16   Statusword
  [0x0048.0] 0x6061:0x00 0x08 INTEGER8     Modes of operation display
  [0x0049.0] 0x6064:0x00 0x20 INTEGER32    Position actual value
  [0x004D.0] 0x606C:0x00 0x20 INTEGER32    Velocity actual value
  [0x0051.0] 0x6077:0x00 0x10 INTEGER16    Torque actual value
  [0x0053.0] 0x230A:0x00 0x20 INTEGER32    Secondary position value
  [0x0057.0] 0x230B:0x00 0x20 INTEGER32    Secondary velocity value
  [0x005B.0] 0x2401:0x00 0x10 UNSIGNED16   Analog input 1
  [0x005D.0] 0x2402:0x00 0x10 UNSIGNED16   Analog input 2
  [0x005F.0] 0x2403:0x00 0x10 UNSIGNED16   Analog input 3
  [0x0061.0] 0x2404:0x00 0x10 UNSIGNED16   Analog input 4
  [0x0063.0] 0x2702:0x00 0x20 UNSIGNED32   Tuning status
  [0x0067.0] 0x60FD:0x00 0x20 UNSIGNED32   Digital inputs
  [0x006B.0] 0x2704:0x00 0x20 UNSIGNED32   User MISO
  [0x006F.0] 0x20F0:0x00 0x20 UNSIGNED32   Timestamp
  [0x0073.0] 0x60FC:0x00 0x20 INTEGER32    Position demand internal value
  [0x0077.0] 0x606B:0x00 0x20 INTEGER32    Velocity demand value
  [0x007B.0] 0x6074:0x00 0x10 INTEGER16    Torque demand

[... 288 lines cut ...]
    Length: .................. 8
    Control Register: ........ 0x64
    Status Register: ......... 0x00
    Enable byte: ............. 0x01
    SM Type: ................. Process Data Out
  SyncManager SM3
    Physical Startaddress: ... 0x1c00
    Length: .................. 8
    Control Register: ........ 0x20
    Status Register: ......... 0x00
    Enable byte: ............. 0x01
    SM Type: ................. Process Data In

The SM2/SM3 Length should be 35/55 instead of 8/8 or? Is the eeprom not correct?

Here are the wireshark captures of the slaveinfo calls.

[both_slaveinfo_calls.zip](https://github.com/OpenEtherCATsociety/SOEM/files/6715253/both_slaveinfo_calls.zip)

I tried to manually hookup the PDOs without success. It looks like the SDOs cant be written or read.
After adding the slaves parameter into ethercatconfiglist.h the SM2/SM3 config was correct but the SDO communication is still not working. 

Any tips how to get the fault slave initialized would be appreciated.

Best regards

ArthurKetels, 2021-06-25:
You are not the first one experiencing this. This is a problem with a number of EtherCAT slaves. The mailbox handler in the slave firmware gets stuck. Probably a bug in the firmware. Resetting the slave to init should clear all internal state of the slave EtherCAT stack. This clearly is not implemented correctly. But there is nothing SOEM can do about it. You have to take this up with the slave vendor.

vargajanko, 2021-06-25:
Thank you for the response. I will get in touch with the manufactur.

closed 2021-06-25 by vargajanko