OpenEtherCATsociety/SOEM #527: Not all slaves reached operational state

jjrbfi, 2021-07-02:
Hello everyone,

I have compile SOEM/RTT_SOEM and works fine.
When I plug the **EL7342** I got those messages (I'm making drivers for it, but I can't test without fixing this error. I got the same error message with or without drivers.):

## Part of the output:

```
0.309 [ Info   ][Master] ec_init on enx98ded01a2af5 succeeded.
0.717 [ Info   ][Master] 14 slaves found and configured.
0.717 [ Info   ][Master] Request pre-operational state for all slaves
0.718 [ Warning][Master] Could not create driver for EK1100
0.718 [ Info   ][Master] Created driver for EL1008, with address 4098
0.718 [ Info   ][Master] Put configured parameters in the slaves.
0.718 [ Info   ][Master] Created driver for EL1008, with address 4099
0.718 [ Info   ][Master] Put configured parameters in the slaves.
0.718 [ Info   ][Master] Created driver for EL1008, with address 4100
0.718 [ Info   ][Master] Put configured parameters in the slaves.
0.719 [ Info   ][Master] Created driver for EL1008, with address 4101
0.719 [ Info   ][Master] Put configured parameters in the slaves.
0.719 [ Info   ][Master] Created driver for EL2008, with address 4102
0.719 [ Info   ][Master] Put configured parameters in the slaves.
0.719 [ Info   ][Master] Created driver for EL2008, with address 4103
0.719 [ Info   ][Master] Put configured parameters in the slaves.
0.719 [ Info   ][Master] Created driver for EL7342, with address 4104
0.719 [ Info   ][Master] Put configured parameters in the slaves.
0.719 [ Info   ][Master] Created driver for EL7342, with address 4105
0.719 [ Info   ][Master] Put configured parameters in the slaves.
0.719 [ Warning][Master] Could not create driver for EL9576
0.719 [ Warning][Master] Could not create driver for EL6900
0.719 [ Warning][Master] Could not create driver for EL1904
0.719 [ Warning][Master] Could not create driver for EL2904
0.719 [ Warning][Master] Could not create driver for EL9505
1.699 [ Info   ][ScriptingService] Request safe-operational state for all slaves
1.707 [ Info   ][ScriptingService] Safe operational state reached for all slaves.
1.708 [ Info   ][ScriptingService] Request operational state for all slaves
3.709 [ ERROR  ][ScriptingService] Not all slaves reached operational state.
3.709 [ ERROR  ][ScriptingService] Slave 1 State= 2 StatusCode=0 : No error
3.709 [ ERROR  ][ScriptingService] Slave 2 State= 2 StatusCode=0 : No error
3.709 [ ERROR  ][ScriptingService] Slave 3 State= 2 StatusCode=0 : No error
3.709 [ ERROR  ][ScriptingService] Slave 4 State= 2 StatusCode=0 : No error
3.709 [ ERROR  ][ScriptingService] Slave 5 State= 2 StatusCode=0 : No error
3.709 [ ERROR  ][ScriptingService] Slave 6 State= 2 StatusCode=0 : No error
3.709 [ ERROR  ][ScriptingService] Slave 7 State= 2 StatusCode=0 : No error
3.709 [ ERROR  ][ScriptingService] Slave 8 State= 2 StatusCode=0 : No error
3.709 [ ERROR  ][ScriptingService] Slave 9 State= 2 StatusCode=0 : No error
3.710 [ ERROR  ][ScriptingService] Slave 10 State= 2 StatusCode=0 : No error
3.710 [ ERROR  ][ScriptingService] Slave 11 State= 2 StatusCode=0 : No error
3.710 [ ERROR  ][ScriptingService] Slave 12 State= 2 StatusCode=0 : No error
3.710 [ ERROR  ][ScriptingService] Slave 13 State= 2 StatusCode=0 : No error
3.710 [ ERROR  ][ScriptingService] Slave 14 State= 2 StatusCode=0 : No error
3.711 [ Debug  ][ScriptingService] Could not determine sample size for type /soem_beckhoff_drivers/DigitalMsg
3.711 [ Debug  ][DI] Creating ROS publisher for port Master.bits on topic DI
3.715 [ Info   ][Thread] Creating Thread for scheduler=ORO_SCHED_OTHER, priority=1, CPU affinity=0, with name='RosPublishActivity'

```
As you can see I got:
```
1.699 [ Info   ][ScriptingService] Request safe-operational state for all slaves
1.707 [ Info   ][ScriptingService] Safe operational state reached for all slaves.
1.708 [ Info   ][ScriptingService] Request operational state for all slaves
```

## What I have tried:
I tried to see the status with Twincat3 (is in OP) and I can change the status without any issue to others...

When I unplug the **EL7342** and run again the master. It detects all the other terminals without any ERROR message and launch fine the master.

Regarding the Debug warning: _Could not determine sample size for type /soem_beckhoff_drivers/DigitalMsg_
I saw that comes from orocos-toolchain.

## Slave info in case needed:

```
Slave:8
 Name:EL7342
 Output size: 160bits
 Input size: 192bits
 State: 4
 Delay: 994[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1008
 Man: 00000002 ID: 1cae3052 Rev: 00140000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:  20 F:00010024 Type:3
 SM3 A:1200 L:  24 F:00010020 Type:4
 FMMU0 Ls:00000002 Ll:  20 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000047 Ll:  24 Lsb:0 Leb:7 Ps:1200 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 27 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 140[mA]
 only LRD/LWR:0

```

## Eepromtool output: 

```
$ rosrun soem eepromtool enx98ded01a2af5 8 -i

SOEM (Simple Open EtherCAT Master)
EEPROM tool
ec_init on enx98ded01a2af5 succeeded.
14 slaves found.
Slave 8 data
 PDI Control      : 0405
 PDI Config       : CC03
 Config Alias     : 0000
 Checksum         : 003D
   calculated     : 003D
 Vendor ID        : 00000002
 Product Code     : 1CAE3052
 Revision Number  : 00140000
 Serial Number    : 00000000
 Mailbox Protocol : 000C
 Size             : 000F = 2048 bytes
 Version          : 0001
End, close socket
End program
```

Any hint for those errors?
Thanks! :)

ArthurKetels, 2021-07-02:
Post a wireshark capture (.pcap file zipped) of the configuration that went wrong. Could be as simple as a wrong timeout.

jjrbfi, 2021-07-05:
Hello, thanks for the fast reply.

Here goes the .pcap file zipped.

[issue_el7342.pcap.zip](https://github.com/OpenEtherCATsociety/SOEM/files/6762389/issue_el7342.pcap.zip)

ArthurKetels, 2021-07-05:
The current problem is clear. In pre-OP there is only one PDO exchange. This triggers the watchdog in all slaves. Then the transition to OP is requested and the code waits for all slaves to have transitioned to OP. Some slaves take more than 100ms to do that. But 100ms is the watchdog time-out for PDO exchange. All slaves that have transitioned to OP must receive PDO updates regularly. 

In the wireshark trace this mechanism is easy to follow. Almost exactly 100ms after OP request one or more slaves trigger the PDO watchdog.

The solution is simple. Keep sending PDO updates while waiting for all slaves to go to OP.

jjrbfi, 2021-07-05:
So, in order to do that, should I modify the delay time in some part of the code or through Twincat?
To keep sending PDO updated while waiting for all slaves?

ArthurKetels, 2021-07-05:
Have a look at the red_test.c example in the test folder. It has a separate RT-task that does the PDO transfers. This is triggered by the configuration routine after request for OP state.

jjrbfi, 2021-07-12:
Hi @ArthurKetels ,

I have been trying to get that working, but unfortunately I haven't.
I been reading quite many "issues" in the repository as well red_test.c to get more knowledge about how code works.

I guess the closed part of the code I have changed to get close to get sending PDO, was:

- **EC_TIMEOUTSTATE**, in order to keep waiting to get the state changed in this line. Or that was the idea I had.
- Changed 100ms to more, to do some tests **SyncDelay**: https://github.com/OpenEtherCATsociety/SOEM/blob/master/soem/ethercatdc.c#L24
- Changed also to more to test **EC_TIMEOUTSTATE**: https://github.com/OpenEtherCATsociety/SOEM/blob/master/soem/ethercattype.h#L76

Also thinking maybe I have to implement this part in the rtt_soem. Not sure if really nedded: 
https://github.com/OpenEtherCATsociety/SOEM/blob/master/test/linux/red_test/red_test.c#L160-L309

Regards.

jjrbfi, 2021-07-19:
Solved using while to reach the state.

Thanks

closed 2021-07-19 by jjrbfi

zouzhe1, 2023-04-18:
@jjrbfi I have encountered a similar issue. May I ask which part of it did you modify?

jjrbfi, 2023-04-18:
> @jjrbfi I have encountered a similar issue. May I ask which part of it did you modify?

Hi, I worked on this couple of years a go. I didn't remember well what was the changed I did.

I think the changes were in:

**soem_master/src/soem_master_component.cpp**

`			// send one valid process data to make outputs in slaves happy
			ec_send_processdata();

                ec_receive_processdata(EC_TIMEOUTRET);

            ec_writestate(0);
            int chk=200; 
            // Wait for all slaves to reach OP state
            do
             {
                ec_send_processdata();
                ec_receive_processdata(EC_TIMEOUTRET);
                //ec_writestate(0);
                ec_statecheck(0, EC_STATE_OPERATIONAL, EC_TIMEOUTRET);
             }
            while (chk-- && (ec_slave[0].state != EC_STATE_OPERATIONAL));
            while (EcatError)
                {
                    log(Error) << ec_elist2string() << endlog();`
                    
                    
Hope it helps your.

jjrbfi, 2023-04-18:
JFYI @zouzhe1 :
I have new terminals drives created since that time. But didn't release due terms and conditions/license this repository have.

I switched to another EtherCAT master with GPL license.