OpenEtherCATsociety/SOEM #528: Question: Changing OP --> SafeOP timeout?

johnneren, 2021-07-05:
Hi.

I was hoping someone could point me in the right direction.
I'm working with a Linux OS which seems to not be that real time stable and I sometimes get > 50ms judder.

My EtherCAT loop is running every 50ms with spikes where it does not send process data for ~120ms.
These spikes causes all the slaves to do a OP -->  SafeOP + ERROR --> OP transition.
I don't need real time data from my ethercat / beckhoff setup and a new value every 500ms is more than enough.

Does anyone know how to increase this OP --> SafeOP 'no process data' watchdog timeout of 100ms on the slaves through SOEM?
I couldn't quite find anything with google or in the example code.

Any feedback would be appreciated :)

ArthurKetels, 2021-07-05:
If you have 120ms latency spikes something is wrong with your Linux configuration. I would try to solve those issues first. Even on RPi systems with Ethernet over USB you can do a few ms maximum latency spikes.

Changing the PDO watchdog in slaves is most often not possible. Some slaves have this as a writeable parameter, but then it is mentioned in the datasheet.

As with most problems, solve the root cause instead of trying to duct-tape over it.

johnneren, 2021-07-05:
Fair enough.

Since I didn't need real time performance I thought this was an option to raise the timeout.
I'm working on a platform with multiple applications written by multiple teams running on the same platform so perhaps one of the other applications are causing the spikes. (I'm getting them even at SCHED_FIFO prio 80)

I'll try and look into that or just keep the SOEM slaves running in PreOP or SafeOP since I only need to read the value of 2 analog inputs every second for logging purpose.

closed 2021-07-05 by johnneren