OpenEtherCATsociety/SOEM #529: EoE: ecx_EOEsetIp returns false altough it succeeded (with EL6601)

Apollo3zehn, 2021-07-06:
I am testing EoE with the EL6601 and have difficulties to set it up correctly using `ecx_EOEsetIp`. My setup is:
PC (NIC 1) -> EK1100 -> EL6601 (with 1 Ethernet Port) -> PC (NIC 2)

Most of the time, when I configure the IP settings over EoE with this setup, the method `ecx_EOEsetIp` returns false. And in Wireshark I can see the `Init Req` but no `Init Res` (see screenshot below). When I rerun `ecx_EOEsetIp`, Wireshark shows me the missing `Init Res` and the new `Init Req`, again without following `Init Res`.

I think the reason is that in the meantime, the EL6601 received other Ethernet frames over its port which are sitting in the mailbox and are getting read by the master instead of the expected `Init Res`. 

When I pull the Ethernet cable and clear the Mailbox (until all buffered frames are processed), the method `ecx_EOEsetIp` returns always `true`.

I guess the reason can be found [here](https://github.com/OpenEtherCATsociety/SOEM/blob/master/soem/ethercateoe.c#L154). That line expects a frame of type `EOE_INIT_RESP` and if it doesn't get it, the method returns unsuccessfully.

[image]

nakarlsson, 2021-07-08:
Ok, you might be on to something. I’ll have a look when the vacations are over.

nakarlsson, 2021-07-08:
Btw, have you checked whats happens in the MBX receive routine, do you have a hook that is suppose to receive data?

Apollo3zehn, 2021-07-08:
No I did not debug into that routine since I was pretty sure that the check `EOE_HDR_FRAME_TYPE_GET(frameinfo1) != EOE_INIT_RESP` in `ecx_EOEsetIp` is checking against the wrong EoE response message (if I interpret the Wireshark output correctly). 

Is there something specific I could in the MBX receive routine?

I do not have any hooks registered.

nakarlsson, 2021-07-08:
Try adding a hook and let it handle other frames, like ARPs and so on.

 `
              /* All non fragment data frame types are expected to be handled by
               * slave send/receive API if the EoE hook is set
               */

             if (EOE_HDR_FRAME_TYPE_GET(frameinfo1) == EOE_FRAG_DATA)
`

nakarlsson, 2021-08-10:
Have you been able to test adding the hook?