OpenEtherCATsociety/SOEM #531: Delay required between ecx_outframe() and ecx_inframe() of nicdrv.c

DamonTTSA, 2021-07-09:
Good day

I have encountered a strange issue in my implementation of SOEM.
After an Ethernet frame has been transmitted by ecx_outframe(), I require a small delay before ecx_inframe() is called.
This delay must be approximately 80us for a relatively reliable result for PDOs and SDOs.

If I do not include this delay, ecx_inframe() searches for an idx that is one larger than the idxf (the found idx from ecx_recvpkt).
"Strange things happened" is then printed, since the idxf is not being waited for anymore.
The returned frames are thus never matched with their IDs, so communication is not possible.
If I decrease this delay to about 40us, PDO communication still works, but SDO communication does not.

I had to implement my own nicdrv.c for my use, but I used the linux example as a basis.

If someone has an idea about where I went wrong or what I'm missing, I would really appreciate the help!

ArthurKetels, 2021-07-09:
This could have multiple causes. 

One that comes to mind is receive packet coalescing. The NIC driver waits releasing a received frame until a next one comes in or a timeout happens. 
SOEMtx(1) -> NICtx(1)
....packet over the wire....
NICrx(1) ... hold in buffer ...
SOEMrx(1) -> no response -> timeout -> clear SOEM buffer (1)
SOEMtx(2) -> NICtx(2)
...packet over the wire....
NICrx(2) -> release packet (1) and (2)
SOEMrx(2) -> get NIC buffer (1) -> no active SOEM buffer(1) found -> strange things have happened

Further look into your mutex locking. Timing routines accuracy. 

It would help if you printed out the events (tx idx, rx idx, rx idxf). Put them in a buffer and print out after lets say 20 events. Printing them immediately would screw with your timing.

An old issues I once had was a NIC driver that echoed the tx packets to rx. You can detect this looking at the MAC address in the packet.

DamonTTSA, 2021-07-12:
Hi @ArthurKetels 

Thank you for your response!
I believe I may have found the problem, but I'd like to ask for some clarification to be sure.

My version of ecx_recvpkt() of nicdrv.c returns true whenever a packet is detected.
In my case, this remains true, even once a packet has been read.
So, for example, if I receive ID 1 and then send ID 2, when I call ecx_recvpkt() again, it will return true, with a packet ID of 1.

I would just like to confirm that ecx_recvpkt() should ONLY return > 0 if a completely NEW packet is detected?

Kind regards

ArthurKetels, 2021-07-12:
This is correct.

DamonTTSA, 2021-07-13:
Great, thank you @ArthurKetels.
The issue I described above was indeed the problem.
Thank you once again for your assistance!

Kind regards

closed 2021-07-13 by DamonTTSA