OpenEtherCATsociety/SOEM #532: calls to timeBeginPeriod and timeEndPeriod do not match (win32)

yschmid-levi, 2021-07-13:
According to the [Microsoft's documentation](https://docs.microsoft.com/en-us/windows/win32/api/timeapi/nf-timeapi-timebeginperiod), calls to timeBeginPeriod and timeEndPeriod must have matching arguments:

> You must match each call to timeBeginPeriod with a call to timeEndPeriod, specifying the same minimum resolution in both calls.

But they don't:

https://github.com/OpenEtherCATsociety/SOEM/blob/219670218b78acf4e4a96da70d36eeb9fc7f96ca/osal/win32/osal.c#L20

https://github.com/OpenEtherCATsociety/SOEM/blob/219670218b78acf4e4a96da70d36eeb9fc7f96ca/oshw/win32/nicdrv.c#L164

ArthurKetels, 2021-07-13:
Yes. you are correct. And the fix is straightforward. However after reading the Microsoft documentation I found a more disturbing announcement.

> Prior to Windows 10, version 2004, this function affects a global Windows setting. For all processes Windows uses the lowest value (that is, highest resolution) requested by any process. Starting with Windows 10, version 2004, this function no longer affects global timer resolution. For processes which call this function, Windows uses the lowest value (that is, highest resolution) requested by any process. For processes which have not called this function, Windows does not guarantee a higher resolution than the default system resolution.
> 
> Starting with Windows 11, if a window-owning process becomes fully occluded, minimized, or otherwise invisible or inaudible to the end user, Windows does not guarantee a higher resolution than the default system resolution. See SetProcessInformation for more information on this behavior.

The first paragraph makes it clear that the fix does not matter any more for Win10 v2004+. The higher resolution is only applied for the process that requests it. When the process ends, the higher time resolution is removed also. And timeEndPeriod is called in SOEM when the program ends. 

The second paragraph is more worrying, the time resolution will drop automatically when the window is no longer visible. This will end the practical use of SOEM on Windows platform for anything requiring real-time behaviour. I must see if there are any new API to get decent timing performance out of Windows 11.

closed 2022-01-13 by ArthurKetels

rvangaal, 2023-02-03:
Found anything yet? This is indeed a big thing in Windows 11.

PR #533 (2022-01-13): matching call to timeEndPeriod and timeBeginPeriod
fixes  #532