OpenEtherCATsociety/SOEM #535: [question] ec_config_overlap_map usage for TI ESC

leadermik, 2021-07-15:
Thank you for you amazing lib! 👏🏿 

I am currently writing my Bachelor Thesis and I am trying to port your stack on the[ Teensy 4.1 board
](https://www.pjrc.com/store/teensy41.html) using its native ethernet controller.  When it is finished I would gladly open source the project as I am mainly using your amazing lib. 

I am trying to communicate with 2 [TI ESC Board](https://www.ti.com/tool/TMDSCNCD28388D) yet with no luck. First board remains in PREOP & the second (on the chain) return an AL_Status code of 0x001E => Invalid Input Configuration. However if I connect the 2 Board to TwinCAT everything works fine. 😓 

On the master side I am calling the `ec_config_init ` and then the `ec_config_map` . So as simple as possible. Packets are send out (wireshark), however I cannot get from PREOP => SAFEOP. 🤔 

I saw this [comment](https://github.com/OpenEtherCATsociety/SOEM/blob/219670218b78acf4e4a96da70d36eeb9fc7f96ca/soem/ethercatconfig.c#L1640) of yours. Should `ec_config_overlap` be used instead of `ec_config_init ` for the TI Slave that I mentioned?  Can you give some more explainations? 🙏🏿 

Thanks again! 🏅

nakarlsson, 2021-07-15:
Yes, you should use the Overlap API for TI ESCs. However, that will not have any affect until sending process data using the LRW commmand.

So, please post the output from slaveinfo -map including a Wireshark.

leadermik, 2021-07-16:
Thanks for very fast answer. I am suspecting that is is some kind of beginner mistake, therefore I dug even more on the stack ( also in the one the TI ESC), yet with little luck 😞 

👉🏿 So here the [slaveinfo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/6833056/slaveinfo.txt) and corresponding [slaveinfo_dump_wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/6833065/slaveinfo_dump_wireshark.zip)

So as you can see I am having 1 Master (Teensy 4.1) and two slaves (TI ESC). The project will have 12 TI ESC at the end. The project is a distributed High Voltage Invertar and the slaves must exchange data as fast as possible.

👉🏿 Here also the SSC (Slave Stack Code) excels with the informations for the ESCs: 
[F2838x CPU1 EtherCAT Slave Integration_Slave0.xlsx](https://github.com/OpenEtherCATsociety/SOEM/files/6833073/F2838x.CPU1.EtherCAT.Slave.Integration_Slave0.xlsx)

[F2838x CPU1 EtherCAT Slave Integration_Slave1.xlsx](https://github.com/OpenEtherCATsociety/SOEM/files/6833075/F2838x.CPU1.EtherCAT.Slave.Integration_Slave1.xlsx)
 
The strange thing is that, if I connect them seperatly with the master, the communication works great. Yet if I daisy chain them, then the second ESC return an AL_Status Code: 0x001e (Invalid input configuration), while the file remains at PREOP from the transmittion PREOP=>SAFEOP

The whole problems happens on this custom piece of code (it is a combination of your examples actually):

`// writestate to all slaves, block until state is reachedd
void cp_ECAT_writestate(ec_state requested_state)
{
    INFO_PRINTF("Request %s state\n", utils_ECAT_state_2_string(requested_state));
    ec_slave[0].state = requested_state;
    do
    {
        // Try to write state
        ec_writestate(0);

        // TODO: Potentially dangerous if wrong data in PDOs
        // Transitioning to OP State requeres one exchange of PDO Data
        if (requested_state == EC_STATE_OPERATIONAL)
        {
            ec_send_processdata();
            ec_receive_processdata(EC_TIMEOUTRET);
        }

        uint16 _actual_state = ec_statecheck(0, requested_state, EC_TIMEOUTSTATE * 4);
        if (_actual_state != requested_state)
        {
            ec_readstate();
            for (uint i = 1; i <= ec_slavecount; i++)
            {
                if (ec_slave[i].state != requested_state)
                {
                    INFO_PRINTF("---> Slave %d State=0x%2.2x toText:%s StatusCode=0x%4.4x : %s\n",
                                i, ec_slave[i].state,
                                utils_ECAT_state_2_string(_actual_state),
                                ec_slave[i].ALstatuscode,
                                ec_ALstatuscode2string(ec_slave[i].ALstatuscode));
                }
            }
        }
        if (EcatError)
            INFO_PRINTF("--> %s \n", ec_elist2string());

    } while (ec_readstate() != requested_state);
    INFO_PRINTF("State %s reached \n", utils_ECAT_state_2_string(requested_state));
    while (EcatError)
        INFO_PRINTF("---> %s \n", ec_elist2string());
}
`
👉🏿 Here you take a look what is actually happening. Everything looks fine ( I think 😅 ). 
[custom_ECAT_writestate_dump_wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/6833109/custom_ECAT_writestate_dump_wireshark.zip)

If I understand it correctly slave are started and configured until # 1160 . Then from # 1160 => # 1169 the states are changed until the error happens then they default back to PREOP. The first ESC also remains there and the second is blinking his error LED. 

👉🏿 Finally I am getting the sam error with TWINcat. So maybe I am making a trivial error or there is something wrong with the SSC

[image]

In any case I am very gratefull for your help, software and support! Your are amazing 🙏🏿 👏🏿

ArthurKetels, 2021-07-16:
In slaveinfo_dump_wireshark.zip both TI slaves actually go to SAFE-OP perfectly OK. Also from the text output you can see:

> Slave:1
>  Name:VerseSlave_Dev04
>  Output size: 128bits
>  Input size: 128bits
>  State: 4

State 4 being SAFE-OP.  You can see this in line 814 and 816 of the wireshark capture.

In your custom code something strange happens. The slave mapping goes fine and then you force all slaves to INIT (line 1161). Then command PRE-OP and then SAFE-OP. The last transition is refused by the slaves. Most likely the slave configuration is reset after the switch to INIT.

Two things, follow the normal flow of state transitions, and make sure the slave firmware does not throw away it's configuration on transition to INIT. At least the master also has to reset the SM and FMMU and SYNC (when applicable) before the slave resets.

I would suggest to first try the standard simple_test code. See if it takes the slaves to OP.

leadermik, 2021-07-18:
Thanks for your advice 👏🏿 

👉🏿 Here [simple_test_dump_wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/6836381/simple_test_dump_wireshark.zip)
 the wireshark output of the simpletest also here the code:
 
`     int i, j, oloop, iloop, chk;
    needlf = FALSE;
    inOP = FALSE;

    for (;;)
    {
        INFO_PRINTF("Starting simple test\n");

        /* find and auto-config slaves */

        if (ec_config_init(FALSE) > 0)
        {
            INFO_PRINTF("%d slaves found and configured.\n", ec_slavecount);

            ec_config_overlap_map(&IOmap);

            ec_configdc();

            INFO_PRINTF("Slaves mapped, state to SAFE_OP.\n");
            /* wait for all slaves to reach SAFE_OP state */
            ec_statecheck(0, EC_STATE_SAFE_OP, EC_TIMEOUTSTATE * 4);

            oloop = ec_slave[0].Obytes;
            if ((oloop == 0) && (ec_slave[0].Obits > 0))
                oloop = 1;
            if (oloop > 8)
                oloop = 8;
            iloop = ec_slave[0].Ibytes;
            if ((iloop == 0) && (ec_slave[0].Ibits > 0))
                iloop = 1;
            if (iloop > 8)
                iloop = 8;

            INFO_PRINTF("segments : %d : %d %d %d %d\n", ec_group[0].nsegments, ec_group[0].IOsegment[0], ec_group[0].IOsegment[1], ec_group[0].IOsegment[2], ec_group[0].IOsegment[3]);

            INFO_PRINTF("Request operational state for all slaves\n");
            expectedWKC = (ec_group[0].outputsWKC * 2) + ec_group[0].inputsWKC;
            INFO_PRINTF("Calculated workcounter %d\n", expectedWKC);
            ec_slave[0].state = EC_STATE_OPERATIONAL;
            /* send one valid process data to make outputs in slaves happy*/
            ec_send_overlap_processdata();
            ec_receive_processdata(EC_TIMEOUTRET);
            /* request OP state for all slaves */
            ec_writestate(0);
            chk = 200;
            /* wait for all slaves to reach OP state */
            do
            {
                ec_send_overlap_processdata();
                ec_receive_processdata(EC_TIMEOUTRET);
                ec_statecheck(0, EC_STATE_OPERATIONAL, 50000);
            } while (chk-- && (ec_slave[0].state != EC_STATE_OPERATIONAL));
            if (ec_slave[0].state == EC_STATE_OPERATIONAL)
            {
                INFO_PRINTF("Operational state reached for all slaves.\n");
                inOP = TRUE;
                /* cyclic loop */
                for (i = 1; i <= 10000; i++)
                {
                    ec_send_overlap_processdata();
                    wkc = ec_receive_processdata(EC_TIMEOUTRET);

                    if (wkc >= expectedWKC)
                    {
                        INFO_PRINTF("Processdata cycle %4d, WKC %d , O:", i, wkc);

                        for (j = 0; j < oloop; j++)
                        {
                            INFO_PRINTF(" %2.2x", *(ec_slave[0].outputs + j));
                        }

                        INFO_PRINTF(" I:");
                        for (j = 0; j < iloop; j++)
                        {
                            INFO_PRINTF(" %2.2x", *(ec_slave[0].inputs + j));
                        }
                        INFO_PRINTF(" T:%" PRId64 "\r", ec_DCtime);
                        needlf = TRUE;
                    }
                    osal_usleep(5000);
                }
                inOP = FALSE;
            }
            else
            {
                INFO_PRINTF("Not all slaves reached operational state.\n");
                ec_readstate();
                for (i = 1; i <= ec_slavecount; i++)
                {
                    if (ec_slave[i].state != EC_STATE_OPERATIONAL)
                    {
                        INFO_PRINTF("Slave %d State=0x%2.2x StatusCode=0x%4.4x : %s\n",
                                    i, ec_slave[i].state, ec_slave[i].ALstatuscode, ec_ALstatuscode2string(ec_slave[i].ALstatuscode));
                    }
                }
            }
            INFO_PRINTF("\nRequest init state for all slaves\n");
            ec_slave[0].state = EC_STATE_INIT;
            /* request INIT state for all slaves */
            ec_writestate(0);
        }
        else
        {
            INFO_PRINTF("No slaves found!\n");
        }
        INFO_PRINTF("End simple test, close socket\n");
        /* stop SOEM, close socket */
        ec_close();
    }
`
I tried it with ec_config_map & ec_config_overlap_map (ec_send_processdata, ec_send_overlap_processdata) respectively, but no diffenrence.

Here the serial output:
` 
config_board_soft done
config_board_hard done
Starting simple test
2 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 38 0 0 0
Request operational state for all slaves
Calculated workcounter 5
Not all slaves reached operational state.
Slave 2 State=0x12 StatusCode=0x001e : Invalid input configuration

Request init state for all slaves
End simple test, close socket
Starting simple test
`

So as you see I get the same AL_Status Erro Code 😞 

1. I will examine the the wireshark output of the last attempt and try to correlate with the code. 
2. I will also try to upgrade the SOEM ECAT lib. Maybe I have an older version. 
3. Finally I try to upload the SOEM modified code here on github so that you can also take a look if you want.

Thanks again

ArthurKetels, 2021-07-18:
I am very happy you have identified the steps to go forward. All 3 steps are worthwhile. 
Always use the latest version here and pull from master branch. Setting up a fork on github with your modified code allows us to much easier spot software mistakes.

One thing I noticed is that the configuration setting of SOEM for slaveinfo and (modified)simple_test are different. Never do that without good reason. 
In ethercatmain.h you modified:
```
/** define maximum number of concurrent threads in mapping */
#define EC_MAX_MAPT           1
```
To a value higher than 1. This is not recommended. There are some bugs there that might bite you. Keep EC_MAX_MAPT at 1. 

It is clear that the SM2/3 configuration is wrong. The calculated sizes are different than what is returned from the slave mapping. And that WILL trigger a status code 0x001e. In contrast, your first program slaveinfo did it perfectly, all SM are in agreement with the mapping.

leadermik, 2021-07-18:
You are true God! 🤩 and you must made my day 😌! That was the bug! 

I must have done it by accident. So unnessary 👎🏿 

I started from [this](https://github.com/lipoyang/SOEM4Arduino) project here and slowly ported the software to Teensy 4.1 board. First I had an external W5500 board over SPI. Then I ported it to its native ethernet driver and now I connected 2 slaves. I do not know what version was used. 

1. I will clean up the code
2. Fork you project
3. Make a pull request

I will also post more code and examples and my final thesis (later)! This would not have been possible without your help 🙏🏿 

Please do not close this issue yet, as minor problem may arise again. 🏅

nakarlsson, 2021-07-19:
I suggest you close it anyway, write a new Issue if/when a new problem arise. We'll still be here.

closed 2021-08-03 by leadermik