OpenEtherCATsociety/SOEM #538: All ec_slave.inputs changed after ec_config_overlap_map_group() despite grouping

Wanga, 2021-08-04:
Hi All,

I'm using both overlap and non-overlap PDO under the same SOEM instance by
1. Change EC_MAXGROUP to 3
2. Assign slave with blockLRW flag = 0 to group 1
Assign slave with blockLRW flag = 1 to group 2
2. `ec_config_overlap_map_group (pIOmap_overlap, 1)`
3. `ec_config_map_group (pIOmap, 2)`

**Issue:** After step 2, slavelist[slave].inputs of all slaves becomes a non-zero value, then in step 3, [this part](https://github.com/OpenEtherCATsociety/SOEM/blob/04f7baec68610093ca379af364b48b9d1ee1360a/soem/ethercatconfig.c#L1010-L1030) will not be executed.

**Cause:**  [This part](https://github.com/OpenEtherCATsociety/SOEM/blob/04f7baec68610093ca379af364b48b9d1ee1360a/soem/ethercatconfig.c#L1473-L1476) of ecx_config_overlap_map_group should have group checking
```
for (slave = 1; slave <= *(context->slavecount); slave++)
{
+   if (!group || (group == context->slavelist[slave].group))
         context->slavelist[slave].inputs += context->grouplist[group].Obytes;
}
```

nakarlsson, 2021-08-04:
Seems right, good catch. Can you create a PullRequest that can be reviewed and tested?

Wanga, 2021-08-04:
please review this PR

commit be1a2df6 (2021-08-10): Merge pull request #539 from Wanga/master

Add group checking for slave input offset calculation

fixes #538
closed 2021-08-10 by nakarlsson