OpenEtherCATsociety/SOEM #540: [question] How to determine the exact size for IOmap

leadermik, 2021-08-10:
Hello 👋🏿 ,

In my application I am not having so much free memory space and I was wondering if I could adapt the dimension of the IOmap as is was default to char IOmap[4096]. 

How exactly is this IOmap being used? How exactly does the function ec_config_overlap_map work?

Many thanks. 

Here is the output of the slaveinfo, with EC_DEBUG activated:

````
Copy SII slave 2 from 1.
Copy SII slave 3 from 1.
ec_config_map_group IOmap:0x2000b4c0 group:0
 >Slave 1, configadr 1001, state 02
  CoE Osize:1160 Isize:1360
 >Slave 2, configadr 1002, state 02
  CoE Osize:1160 Isize:1360
 >Slave 3, configadr 1003, state 02
  CoE Osize:1160 Isize:1360
     ISIZE:1360 1360 OSIZE:1160
  SM programming
    SM2 Type:3 StartAddr:1100 Flags:00010064
    SM3 Type:4 StartAddr:1400 Flags:00010020
     ISIZE:1360 1360 OSIZE:1160
  SM programming
    SM2 Type:3 StartAddr:1100 Flags:00010064
    SM3 Type:4 StartAddr:1400 Flags:00010020
     ISIZE:1360 1360 OSIZE:1160
  SM programming
    SM2 Type:3 StartAddr:1100 Flags:00010064
    SM3 Type:4 StartAddr:1400 Flags:00010020
  OUTPUT MAPPING
    FMMU 0
      SM2
    slave 1 Outputs 0x2000b4c0 startbit 0
  OUTPUT MAPPING
    FMMU 0
      SM2
    slave 2 Outputs 0x2000b551 startbit 0
  OUTPUT MAPPING
    FMMU 0
      SM2
    slave 3 Outputs 0x2000b5e2 startbit 0
 =Slave 1, INPUT MAPPING
    FMMU 1
      SM3
    Inputs 0x2000b673 startbit 0
 =Slave 2, INPUT MAPPING
    FMMU 1
      SM3
    Inputs 0x2000b71d startbit 0
 =Slave 3, INPUT MAPPING
    FMMU 1
      SM3
    Inputs 0x2000b7c7 startbit 0
IOmapSize 945
3 slaves found and configured.
Calculated workcounter 9

Slave:1
 Name:TI C28xx Slave Devices
 Output size: 1160bits
 Input size: 1360bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
[... 318 lines cut ...]
  [0x0360.0] 0x7006:0x02 0x20 REAL32       Current_Sensor_2
  [0x0364.0] 0x7006:0x03 0x20 REAL32       Current_Sensor_3
  [0x0368.0] 0x7007:0x01 0x20 REAL32       Current_Sensor_1
  [0x036C.0] 0x7007:0x02 0x20 REAL32       Current_Sensor_2
  [0x0370.0] 0x7007:0x03 0x20 REAL32       Current_Sensor_3
  [0x0374.0] 0x7008:0x01 0x20 REAL32       Current_Sensor_1
  [0x0378.0] 0x7008:0x02 0x20 REAL32       Current_Sensor_2
  [0x037C.0] 0x7008:0x03 0x20 REAL32       Current_Sensor_3
  [0x0380.0] 0x7009:0x01 0x20 REAL32       Current_Sensor_1
  [0x0384.0] 0x7009:0x02 0x20 REAL32       Current_Sensor_2
  [0x0388.0] 0x7009:0x03 0x20 REAL32       Current_Sensor_3
  [0x038C.0] 0x700A:0x01 0x20 REAL32       Current_Sensor_1
  [0x0390.0] 0x700A:0x02 0x20 REAL32       Current_Sensor_2
  [0x0394.0] 0x700A:0x03 0x20 REAL32       Current_Sensor_3
  [0x0398.0] 0x700B:0x01 0x20 REAL32       Current_Sensor_1
  [0x039C.0] 0x700B:0x02 0x20 REAL32       Current_Sensor_2
  [0x03A0.0] 0x700B:0x03 0x20 REAL32       Current_Sensor_3
  [0x03A4.0] 0x700C:0x01 0x20 REAL32       Current_Sensor_1
  [0x03A8.0] 0x700C:0x02 0x20 REAL32       Current_Sensor_2
  [0x03AC.0] 0x700C:0x03 0x20 REAL32       Current_Sensor_3
  [0x03B0.0] 0x9000:0x01 0x08 UNSIGNED8    Inverter_ID
[DATA1]
End slaveinfo, close socket

````

nakarlsson, 2021-08-11:
You can read about the IOmap'ing in https://openethercatsociety.github.io/doc/soem/tutorial_8txt.html.

Roughly it equals the number of PDO bytes produced by the slaves.
Say you have a 8bit DI , a 32Bit AI , a 16 Bit AO. It would require a IOmap of minimum (8 + 32 + 16) /  8 = 7 Bytes.

leadermik, 2021-08-11:
Many thanks 👍🏿

closed 2021-08-11 by leadermik