OpenEtherCATsociety/SOEM #541: Not all slaves reached operational state

wengertom, 2021-08-20:
Hello,
I'm trying to control a robot via EtherCAT from my Laptop. I got SOEM up and running and from the beginning it looks all fine. 
After running slaveinfo reserve following output:

````
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eno1 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:netX
 Output size: 288bits
 Input size: 672bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000584 ID: 00001016 Rev: 00000002
 SM0 A:1000 L:1024 F:00010036 Type:1
 SM1 A:1400 L:1024 F:00010032 Type:2
 SM2 A:1800 L:  36 F:00010074 Type:3
 SM3 A:207c L:  84 F:00010030 Type:4
 FMMU0 Ls:00000000 Ll:  36 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000024 Ll:  84 Lsb:0 Leb:7 Ps:207c Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 1024 rd: 1024 MBX protocols : 0e
 CoE details: 0b FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6040:0x00 0x10 UNSIGNED16   
  [0x0002.0] 0x607A:0x00 0x20 INTEGER32    
  [0x0006.0] 0x6840:0x00 0x10 UNSIGNED16   
  [0x0008.0] 0x687A:0x00 0x20 INTEGER32    
  [0x000C.0] 0x7040:0x00 0x10 UNSIGNED16   
  [0x000E.0] 0x707A:0x00 0x20 INTEGER32    
  [0x0012.0] 0x7840:0x00 0x10 UNSIGNED16   
  [0x0014.0] 0x787A:0x00 0x20 INTEGER32    
  [0x0018.0] 0x8040:0x00 0x10 UNSIGNED16   
  [0x001A.0] 0x807A:0x00 0x20 INTEGER32    
  [0x001E.0] 0x8840:0x00 0x10 UNSIGNED16   
  [0x0020.0] 0x887A:0x00 0x20 INTEGER32    
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0024.0] 0x6041:0x00 0x10 UNSIGNED16   
  [0x0026.0] 0x6064:0x00 0x20 INTEGER32    
  [0x002A.0] 0x606C:0x00 0x20 INTEGER32    
  [0x002E.0] 0x60F4:0x00 0x20 INTEGER32    
  [0x0032.0] 0x6841:0x00 0x10 UNSIGNED16   
  [0x0034.0] 0x6864:0x00 0x20 INTEGER32    
  [0x0038.0] 0x686C:0x00 0x20 INTEGER32    
  [0x003C.0] 0x68F4:0x00 0x20 INTEGER32    
  [0x0040.0] 0x7041:0x00 0x10 UNSIGNED16   
  [0x0042.0] 0x7064:0x00 0x20 INTEGER32    
  [0x0046.0] 0x706C:0x00 0x20 INTEGER32    
  [0x004A.0] 0x70F4:0x00 0x20 INTEGER32    
  [0x004E.0] 0x7841:0x00 0x10 UNSIGNED16   
  [0x0050.0] 0x7864:0x00 0x20 INTEGER32    
  [0x0054.0] 0x786C:0x00 0x20 INTEGER32    
  [0x0058.0] 0x78F4:0x00 0x20 INTEGER32    
  [0x005C.0] 0x8041:0x00 0x10 UNSIGNED16   
  [0x005E.0] 0x8064:0x00 0x20 INTEGER32    
  [0x0062.0] 0x806C:0x00 0x20 INTEGER32    
  [0x0066.0] 0x80F4:0x00 0x20 INTEGER32    
  [0x006A.0] 0x8841:0x00 0x10 UNSIGNED16   
  [0x006C.0] 0x8864:0x00 0x20 INTEGER32    
  [0x0070.0] 0x886C:0x00 0x20 INTEGER32    
  [0x0074.0] 0x88F4:0x00 0x20 INTEGER32    
End slaveinfo, close socket
End program
````

But when I try to run simple_test I get stuck at "Not all slaves reached operational state.":
````
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on eno1 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 120 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x04 StatusCode=0x0000 : No error

Request init state for all slaves
End simple test, close socket
End program
````

I'm quite new to SOEM so I tried to find some help in the tutorial.txt and other solved issues. I tried increasing the waiting time to get to OP_State but still same result.
The Problem for me is that the StatusCode and the message "No error" is not giving me any hint what could be going wrong.

I would be thankful for any help. Probably its an easy fix and I'm just not seeing it or readying the wrong closed issues.
Thank you

wengertom, 2021-08-20:
Is it possible a problem, that my SM are not assigned. If I test my laptop with a simple EtherCAT Slave it all works fine. Is it possible to load the ESI file from my robot controller into simple_test? Or is it something else what is causing the Problem?

nakarlsson, 2021-08-21:
Post the ESI and a Wireshark of the startup.

wengertom, 2021-08-23:
Good Morning,

thanks allot for your fast response! 
here are the two wireshark from first running slaveinfo eno1 -map:
[slaveinfo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/7029967/slaveinfo.txt)

and here the one from running simple_test eno1:
[simple_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/7029964/simple_test.txt)

I changed the File type to .txt from .pcapng because of GitHub upload filter.

I'm waiting for the ESI file from the company of the controller. Then I will upload it here as well.

Is it possible to import the ESI file into soem like in TwinCAT? I couldn't find anything regarding implementing an ESI.xml file.

Your help is very much appreciated. 

best regards

wengertom, 2021-08-23:
Hi nakarlsson,
I got the ESI file from the company of the controler now as well.
I changed the ending of the .xml File again because of the support of GitHub. 

ESI File:
[ESI.txt](https://github.com/OpenEtherCATsociety/SOEM/files/7031041/ESI.txt)
 
If something is not working with the files with .txt ending I can also put them in an .zip folder if you prefer that.

Thank you a lot for your help!

ArthurKetels, 2021-08-23:
Thanks for the information, that is very helpful. There is nothing wrong with slaveinfo and simple_test in relation with your slave. The ESI file tells what is missing.

```
						<InitCmd>
							<Transition>PS</Transition>
							<Index>#x60da</Index>
							<SubIndex>0</SubIndex>
							<Data>05000000</Data>
							<Comment>Ch 0 Synch Function Settings</Comment>
						</InitCmd>
						<InitCmd>
							<Transition>PS</Transition>
							<Index>#x68da</Index>
							<SubIndex>0</SubIndex>
							<Data>05000000</Data>
							<Comment>Ch 1 Synch Function Settings</Comment>
						</InitCmd>
						<InitCmd>
							<Transition>PS</Transition>
							<Index>#x70da</Index>
							<SubIndex>0</SubIndex>
							<Data>05000000</Data>
							<Comment>Ch 2 Synch Function Settings</Comment>
						</InitCmd>
						<InitCmd>
							<Transition>PS</Transition>
							<Index>#x78da</Index>
							<SubIndex>0</SubIndex>
							<Data>05000000</Data>
							<Comment>Ch 3 Synch Function Settings</Comment>
						</InitCmd>
						<InitCmd>
							<Transition>PS</Transition>
							<Index>#x80da</Index>
							<SubIndex>0</SubIndex>
							<Data>05000000</Data>
							<Comment>Ch 4 Synch Function Settings</Comment>
						</InitCmd>
						<InitCmd>
							<Transition>PS</Transition>
							<Index>#x88da</Index>
							<SubIndex>0</SubIndex>
							<Data>05000000</Data>
							<Comment>Ch 5 Synch Function Settings</Comment>
						</InitCmd>
```
You will need to set-up a pre-op to safe-op hook in SOEM (see tutorial.txt). In that slave specific function you have to program the slave parameters according to the list above. For example the last init command is CoE write to index 0x88da subindex 0 with value 0x00000005.

Also according to the ESI file the slave will only operate in synchronized mode. So you have to set SYNC0 to the required time interval and at the same time synchronize your master clock to the reference slave clock. See earlier postings here on how to do that.

```
				<Dc UnknownFRMW="0" Unknown64Bit="0">
					<OpMode>
						<Name>DC-synchronized</Name>
						<Desc>DC for synchronization</Desc>
						<AssignActivate>#x0300</AssignActivate>
						<CycleTimeSync0 Factor="1">0</CycleTimeSync0>
						<ShiftTimeSync0>0</ShiftTimeSync0>
					</OpMode>
				</Dc>
```

Remember that SOEM is only a library, you have to build the master application. And for a beginner you took on a complex job.

wengertom, 2021-08-24:
Hello @ArthurKetels ,
Thank you very much for your effort and your information. First of all I'm happy to hear that nothing is wrong with my slaveinfo/simple_test or neither with my slave. With that information I can keep going and build on what I have so far, that is alredy helping me a lot on my project. 

If I'm getting your answer right for a more complex slave, like my controler, the pre-implemented simple_test can't directly work because of distributed clock and the need of pre-operational state before going into the final operational state. I will check the tutorial again and see what I need to implement. 

_"In that slave specific function you have to program the slave parameters according to the list above. For example the last init command is CoE write to index 0x88da subindex 0 with value 0x00000005."_
1. Am I understanding that right that I have to implement the parameters manuel into my Master for the controler slave? 
2. Every parameters from the ESI file or only the ones you were posting above for the **Ch * synch function**?

That brings me to the question if is somehow possible to load the ESI.xml file into my master application. then for the future project it would mean if my slave parameters change and I get a new ESI file I could just reload the new one and get all the parameters without having to hard implement them (like its possible in TwinCAT for example). But not to sure if I'm thinking to far or if that is not even necessary. Most likely I Need to dig deeper into the tutorial and other solved issues first. Only if you know there is something what makes my search easier I would be really thankful. 

I'm on the implementation of the SOEM library and I got some first tests running with a not so complex slave, but therefore I didn't need to implement any parameters from an ESI file.

Your help is very much appreciated.

best regards

wengertom, 2021-08-30:
Hello @ArthurKetels ,
Thanks again for all your help!

The last days I was readying many closed issues and had a closer look into the tutorial again. With that I managed to answer myself most of the above asked questions. 
 
While I was testing some changes at my master implementation to get the CoE init command for the Synch function settings running. 
I also tried to implement the synchronization of my master clock to the reference slave clock. After using:

`c_dcsync0(1, TRUE, 2000000U, 0);`

I tried running simple_test again and I reached immediate Operational state for all slaves.

So my question now, do I actually need to set-up a pre-OP to safe-OP hook in SOEM?

kind regards

ArthurKetels, 2021-08-30:
Have a look at test/win32/simple_test/simple_test.c It shows how to implement the PO2SOconfig hook.

And, of course, when a slave works just out of the box, don't add any fancy stuff. Good is good enough.

I am also glad that you where able to figure out things for yourself. It is a bit of effort, but the learning experience will enhance your skills a lot more than just copying work of others.

wengertom, 2021-09-01:
Hello, 
thanks a lot for your quick answer. Yeah I found the example for the hook in the tutorial as well. 

The learning experience is great. Takes some time and nerves but when you get one step further its totally worth it. 

Just one question, is it also possible during or right after PO2SO to read Data from the slave with ec_SDOread()? 
Since I get a weird behavior right after PO2SOconfig. While testing I had two ec_SDOread() right after each other trying to read from the same index and the first one was just 0 and second one gave me the expected data.

Thanks again.

wengertom, 2021-09-03:
Will close this one, since my main question is solved.

Thank you guys for all the help! Helped me a lot since I was not sure if my starting setup was fine.  
Since I get now to OP state and already can get some data from my controller I'm getting much closer to get my controller working as I need for my project.

Cheers

closed 2021-09-03 by wengertom