OpenEtherCATsociety/SOEM #550: Asynchronous and synchronous communication?

wengertom, 2021-09-03:
Hello,

I'm progressing with my EtherCAT-Master implementation. I get my EtherCAT to OP-State and can read and write with ec_SDOread() and ec_SDOwrite(). 

But for my understanding this is the asynchronous communication, so I only should use this for startup and bringing my components to OP-State. 
For my cyclic process data loop I need the synchronous communication with slave[i].outputs and slave[i].inputs. 

1. Is this the right understanding, or am I on the wrong way?

But to access the PDOs on that way with inputs and outputs i have some weird behavior. For now I'm only trying to read out the values of my Robot controller (Slave[1]) with a struct Rob_response. 

````
struct Rob_response
{
    uint16  status_word_1;
    int32   position_actual_value_1;
    int32   velocity_actual_value_1;
    int32   following_error_actual_value_1;
    uint16  status_word_2;
    int32   position_actual_value_2;
    int32   velocity_actual_value_2;
    int32   following_error_actual_value_2;
    ...
}

...

           response = (struct Rob_response *)(ec_slave[1].inputs);
           ec_send_processdata();
           wkc = ec_receive_processdata(EC_TIMEOUTRET);

            READ(1, 0x6064, 0, buf32, "Axis 1 Position");
            printf(" pos Axis 1: 0x%x \n", response->position_actual_value_1);

            READ(1, 0x6864, 0, buf32, "Axis 2 Position");
            printf(" pos Axis 2: 0x%x \n", response->position_actual_value_2);

````

The READ function is a ec_SDOread(). 

So my problem is that I don't get the same results. Shouldn't I get the same result? 

This is the Slaveinfo -map

````
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eno1 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:netX
 Output size: 288bits
 Input size: 672bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000584 ID: 00001016 Rev: 00000002
 SM0 A:1000 L:1024 F:00010036 Type:1
 SM1 A:1400 L:1024 F:00010032 Type:2
 SM2 A:1800 L:  36 F:00010074 Type:3
 SM3 A:207c L:  84 F:00010030 Type:4
 FMMU0 Ls:00000000 Ll:  36 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000024 Ll:  84 Lsb:0 Leb:7 Ps:207c Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 1024 rd: 1024 MBX protocols : 0e
 CoE details: 0b FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6040:0x00 0x10 UNSIGNED16   
  [0x0002.0] 0x607A:0x00 0x20 INTEGER32    
  [0x0006.0] 0x6840:0x00 0x10 UNSIGNED16   
  [0x0008.0] 0x687A:0x00 0x20 INTEGER32    
  [0x000C.0] 0x7040:0x00 0x10 UNSIGNED16   
  [0x000E.0] 0x707A:0x00 0x20 INTEGER32    
  [0x0012.0] 0x7840:0x00 0x10 UNSIGNED16   
  [0x0014.0] 0x787A:0x00 0x20 INTEGER32    
  [0x0018.0] 0x8040:0x00 0x10 UNSIGNED16   
  [0x001A.0] 0x807A:0x00 0x20 INTEGER32    
  [0x001E.0] 0x8840:0x00 0x10 UNSIGNED16   
  [0x0020.0] 0x887A:0x00 0x20 INTEGER32    
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0024.0] 0x6041:0x00 0x10 UNSIGNED16   
  [0x0026.0] 0x6064:0x00 0x20 INTEGER32    
  [0x002A.0] 0x606C:0x00 0x20 INTEGER32    
  [0x002E.0] 0x60F4:0x00 0x20 INTEGER32    
  [0x0032.0] 0x6841:0x00 0x10 UNSIGNED16   
  [0x0034.0] 0x6864:0x00 0x20 INTEGER32    
  [0x0038.0] 0x686C:0x00 0x20 INTEGER32    
  [0x003C.0] 0x68F4:0x00 0x20 INTEGER32    
  [0x0040.0] 0x7041:0x00 0x10 UNSIGNED16   
  [0x0042.0] 0x7064:0x00 0x20 INTEGER32    
  [0x0046.0] 0x706C:0x00 0x20 INTEGER32    
  [0x004A.0] 0x70F4:0x00 0x20 INTEGER32    
  [0x004E.0] 0x7841:0x00 0x10 UNSIGNED16   
  [0x0050.0] 0x7864:0x00 0x20 INTEGER32    
  [0x0054.0] 0x786C:0x00 0x20 INTEGER32    
  [0x0058.0] 0x78F4:0x00 0x20 INTEGER32    
  [0x005C.0] 0x8041:0x00 0x10 UNSIGNED16   
  [0x005E.0] 0x8064:0x00 0x20 INTEGER32    
  [0x0062.0] 0x806C:0x00 0x20 INTEGER32    
  [0x0066.0] 0x80F4:0x00 0x20 INTEGER32    
  [0x006A.0] 0x8841:0x00 0x10 UNSIGNED16   
  [0x006C.0] 0x8864:0x00 0x20 INTEGER32    
  [0x0070.0] 0x886C:0x00 0x20 INTEGER32    
  [0x0074.0] 0x88F4:0x00 0x20 INTEGER32    
End slaveinfo, close socket
End program
````

Do I manually need to assign the PDOs for the Slave[1].Input and .Output function? 
I'm wondering since I get some results with response->position_actual_value_1 and with the others as well. But for the first one for example it seems like its only an int16 and not as I signed in the struct an int32.

Would be really glad if someone could tell me if I'm on a wrong way or if something is missing. 

Thanks for every help.

Cheers

ArthurKetels, 2021-09-03:
A standard beginners mistake. You have not defined the input struct as packed. If you do not do this the compiler will optimize each struct entry at an optimal boundary (in most cases 32/64 bits).  For example your first entry is `uint16 status_word_1` , even when the variable is defined as 16 bits in reality this will use 32/64 bits of memory. Therefore you second item is offset by 2/6 bytes in memory. 

Input and output structures that overlay the IOmap must always be packed. Please reread the tutorial.txt carefully, the example uses PACKED. 

And, except for testing your data structures, never use ec_SDOread() or ec_SDOwrite() in the real time loop. CoE mailbox transfers are slooooooow.

wengertom, 2021-09-06:
Thanks a lot for the fast and helpful advice! 

That did the trick for me. 

I orientated my code on an example where there wasn't a packed struct. But now I see the example in the tutorial. I didn't know that it needs to be packed but as you wrote otherwise the compiler makes some unwanted optimization and that explains the weird output I got.

Thank you so much for the great explanation !!!

Yeah I had SDOread and SDOwrite only for testing purpose in my real time loop. Already had connection lost because of to many slow CoE  mailbox transfers... ;)

Cheers

closed 2021-09-06 by wengertom

wengertom, 2021-09-09:
Hello @ArthurKetels 

Thanks to your help with my beginner mistake of the struct I made big progress on my SOEM implementation. 

After adding more and more stuff I realized it would be great to have my PDOs organized in an array since I have on my controller 6 times the same pattern, for example the transmission:

````
typedef struct PACKED 
{
    int16  control_word_1;
    int32   target_position_1;
    uint16  control_word_2;
    int32   target_position_2;
    uint16  control_word_3;
    int32   target_position_3;
    uint16  control_word_4;
    int32   target_position_4;
    uint16  control_word_5;
    int32   target_position_5;
    uint16  control_word_6;
    int32   target_position_6;
}Rob_transmission
````

Do you know if there is a way to save the outputs direct in an array: control_words[i] and target_position[i].

So far I have some functions with different data type implemented witch do the trick. But since I need to read and write the Data from and into the arrays in every cycle loop it is quite some computing effort and I was wondering if there is a much more common way to do this.

Something like this, if that explains my question better:
````
typedef struct PACKED 
{
    int16  control_word[0];
    int32   target_position[0];
    uint16  control_word[1];
    int32   target_position[1];
    ....
}Rob_transmission
````

I couldn't find a good solution. Any help is much appreciated.
Thanks

ArthurKetels, 2021-09-09:
For structs that overlay the IOmap you have to use the same structure as the slave. In your case it would be better if the slave had organised its variables better, i.e. first all control words, then the target positions. This is a case of "don't blame the messenger", SOEM can not help it the slave is less than optimal. 

The most effective solution is to copy the EtherCAT struct into your own struct/array that is not PACKED. Yes it will need some CPU cycles, but be aware that every time you have to access a PACKED struct member special assembly code is generated anyway. So in the end it does not matter much.

wengertom, 2021-09-09:
Hi @ArthurKetels ,

Thanks a lot for your fast reply. Then I will stay with my function copying the values every cycle into arrays. 
I just had another Idea working with pointer arrays. So have an array with  

````
int16  *p_control_word[6];
....

p_control_word[0] = &transmission->control_word_1;
p_control_word[1] = &transmission->control_word_2;

....

````

Then I can cycle through them with a for loop:
````
for(int i=0;i<6;i++)
{
    *p_control_word[i] = 11;
}
````

So I only need to initialize it at the start and then I can access my Data via the array pointer.
Not to sure if that's a "nice" way of accessing the PACKED struct but it works and I don't need to write the data to my array every cycle loop.

ArthurKetels, 2021-09-09:
There is a possible issue with pointers pointing inside a packed struct. GCC now also emits a warning when you try to do that. Depending on target CPU and optimisation settings the compiler might generate code that  relies on correct alignment. When this happens you end up with very hard to debug code that sometimes works and sometimes not. Using a packed struct is guaranteed to work because the compiler knows there is no alignment guarantee. When you remove that knowledge by using pointers that point inside the struct the compiler might do the wrong thing.

Please google the issue of using pointers in packed structs.

wengertom, 2021-09-09:
Ok, thanks. didn't know about that so far.

I will have a look at it. Then I probably should stay with my functions from start and run them every loop.

Thanks for all your help.
Cheers