OpenEtherCATsociety/SOEM #551: Plug-in connectivity with SOEM

eliapellegrino, 2021-09-06:
Dear,
I would like to know if the current version of SOEM supports the plug-in connectivity. **What I mean**, is that I can connect an ESC to an already configured network, without the need of switching off and on the entire system.
In particular, can I call the function *ec_config()* on an already configured network?
Any help would be really appreciate.
Elia

ArthurKetels, 2021-09-06:
Yes, this is possible, but with some restrictions. The ec_config() function will reset the slaves back to init and reprogram the SM and FMMU. As long the slaves do not change their configuration during these actions it will work as-is. 

When slaves need to remain in safe-op or pre-op state to hold their configuration it is possible to plug-in with SOEM but you need to make a special ec_config() like function. Do not reset slaves but try to discover them and figure out their configuration. As long the ec_slave structure is filled with the correct actual slave data, SOEM will be able to just continue where the slaves where.

eliapellegrino, 2021-09-06:
@ArthurKetels great answer, thank you so much!
If I can ask, so actually is not possible to implement such a feature when some ESC are in OPerational state? I mean: imagine we have a control box with a robot, and with the same control box (using a different port) we want to command some other ESCs without stopping the robot. For my understanding this is not possible?

ArthurKetels, 2021-09-06:
The problem with operational state is that there is a watchdog attached (standard 100ms). If there are no SM updates within that time the slave will throw an error and revert to safe-op state. If you can manage the setup within that time you could take over in operational state too. 

You second question is totally different. As you know the communication in EtherCAT forms a ring structure, with all slaves in series in the ring. Default there is one master port that acts as the closure of the ring. One transmit gate and one receive gate combined in one Ethernet port. 

The redundant configuration is the same as the standard but now there are two Ethernet ports involved and after the last slave we connect the second master Ethernet port. We have now created a double ring. The Tx1 traverses the slaves and is received by Rx2. Network port 2 now returns the message over Tx2 which traverses the slaves in the other direction to be received by Rx1. Port 2 just copies Rx2 to Tx2. But it also monitors the traffic. If the ring now breaks, somewhere between two slaves, we get two real rings, and port 2 switches to PDO data for the second ring while port 1 still services the first ring. This all works by virtue of the automatic port bypass of each slave.

The third situation is when you have two separate slave rings to begin with. One ring is served by network port 1 and the other ring by network port 2. The two rings work completely independent. This is possible in SOEM by default.  This is why we introduced the extended API where we use ecx_ instead of ec_. We have added the "context" where we can define as many separate rings as needed (can be tens of rings). Each context has it's own network port associated and has it's own configuration and IOmap space. See tutorial.txt how to handle context.

In your example I would place the robot in ring 1 wile having the other ESC's in ring 2.

eliapellegrino, 2021-09-22:
Dear @ArthurKetels, thank you very much for the support! I will close this issue