OpenEtherCATsociety/SOEM #554: How do i communicatie from soem to an Serial interface

zepandasky, 2021-09-10:
I am currently developing an ethercatmaster using SOEM.
I now have an EL6021 Beckhoff 1-channel EtherCAT Terminal with RS-422/485.

But i am neither able to obtain any information regarding the RX data nor can i send data.
I have taken a look in the datasheet and see i can configure the Serial settings by writing the CoE objects. 
The datasheet talks about sending and receiving the data. But all of the data_out[x] indexes are read-only so i cant write to them.
Anyone have any idea or experience where i can write my data too so I can send data from my ethercatmaster to my EL6021 which then sends out the data over the serial interface to any device connected to it.

ArthurKetels, 2021-09-11:
If you look at the datasheet of the EL6021 it should be clear. The default PDO mapping is 22 bytes input and 22 bytes of output buffer. For management there is a 16bit status and a 16bit command word. They are all mapped in SOEM IOmap and you can access them via the pointer ec_slave[n].inputs and ec_slave[n].outputs (where n is the slave number of the EL6021).

Each PDO cycle the whole buffer and command/status is refreshed. In order to update the slave state machine you have to toggle the command tx/rx bit. 

For example, you write the first 10 bytes of the TX buffer, you write value 10 in CW.8..15. Then you toggle CW.0. The moment SOEM updates the PDO the slave sees you have 10 bytes in the buffer and you want to send them. 10 bytes are being transmitted and the slave toggles SW.0 to indicate it has accepted you tx request. 

To receive bytes you monitor SW8..15, if there are any bytes in the buffer you copy this amount in your program and toggle CW.1 to indicate you have read the bytes. When the slave sees this it acknowledges the read with a toggle of SW.1. The slave can now fill the buffer again from byte 0.

This ping-pong action for tx and rx is a very standard way to sync the serial port with EtherCAT. It is easiest if you implement this protocol in a state-machine running in the real-time PDO task of SOEM. To be clear, you do all this reading and writing in the IOmap, you do not use ec_SDOread() and ec_SDOwrite().

zepandasky, 2021-09-11:
Thanks, that makes sense. So the fact that the outputs pro
Mapping are read only are no problem if you manipulate the io map directly?

closed 2021-09-14 by zepandasky