OpenEtherCATsociety/SOEM #555: EoE example not working

zepandasky, 2021-09-14:
Today i've been busy trying to get the eoe_test.c example working for my EL6601 and EL6614.
Haven't been succesfull yet.
The problem seems to be that the `ecx_EOEsetIp() ` is returning a wkc of -5.
I used a debugger to step through the function and it fails at this point
`      /* read slave response */
      wkc = ecx_mbxreceive(context, slave, (ec_mbxbuft *)&MbxIn, timeout); `
Line 146 in file ethercateoe.c
This returns an wkc of -5. 
Then it goes to function 'int ecx_mbxreceive()' in ethercatmain.c on line 1000
When it goes to 1022 wkc returns a normal positive number. SMstat = 256, 0x100

then it goes to line 1031 but SMstat & 0x08 returns 0 offcourse.
So it returns 0 and goes to 1101, where wkc is being made equal to EC_TIMEOUT.
With as result that line 146 in ethercateoe.c fails.

For me its unclear wheter the example is wrong or its a bug in the library.
Any help is welcome

nakarlsson, 2021-09-14:
Could you provide a wirershark log?

zepandasky, 2021-09-14:
@nakarlsson 
Im currently working on a remote linux machine. So i dont have the ability to use a gui.
i can however use tcpdump?

zepandasky, 2021-09-14:
[image]

Left side is my tcpdump right hand side is my output of the example program.
All i did was in the code change the send and receive function slave numbers to the slave's with the ethernetport in my setup. So for me it would be slave 6 or 17 (both have ethernet capabilities).

zepandasky, 2021-09-14:
I seem to understand that there might be a problem with my perspective of receive and sending.
My ultimate goal is using the ethernet ports as a simple connector for people connecting their laptop. So all the ethercat modules have to do is making sure the ethernet traffic is being transmitted over the network. So the data doesnt have to be sent or received by the master.

nakarlsson, 2021-09-14:
Dave the data from tcpdunp to file and post instead. The startup and some runtime.

nakarlsson, 2021-09-15:
NOTE: eoe_test is a very simple sample EoE test, it doesn't include the actual "TCP/IP" stack bindings, it just give an example of the API. But in your case you seem to have trouble just setting the IP. Try adding the set MAC address option as well, see if the slave respond. Again, please send a pcap.

Next step, in a real life application send/recv fragments should be used together with a TAP of some kind .

zepandasky, 2021-09-15:
[enp3s0.zip](https://github.com/OpenEtherCATsociety/SOEM/files/7168067/enp3s0.zip)
This files is the pcap file for the interface which is connected to all the ethercat slaves.

When i connect my network switch to the other ethernet port which is a different block of io modules i see the eoe example receives a lot of frames of type 0x806, and it only does that when i connect that switch. 

So if i understand correctly if want to do regular TCP/IP routing over my ethercat modules i need to always go through the ethercatmaster which then determines to which slave to send the data to (acting like a router/switch). 
i didnt attach that .pcap files due to possible security issues.

nakarlsson, 2021-09-15:
doesn't look like the slave or wireshark like the setIP. 
Try adding MAC and see what happens?

sample I've used in the past with a SSC slave. Use you own settings.

   uint8 mux_mac[] = { 0x1E, 0x30, 0x6C, 0xA2, 0x45, 0x5E };
   eoe_param_t ipsettings, re_ipsettings;
   memset(&ipsettings, 0, sizeof(ipsettings));
   memset(&re_ipsettings, 0, sizeof(re_ipsettings));

   ipsettings.ip_set = 1;
   ipsettings.subnet_set = 1;
   ipsettings.default_gateway_set = 1;
   ipsettings.mac_set = 1;

   EOE_IP4_ADDR_TO_U32(&ipsettings.ip, 192, 168, 9, 200);
   EOE_IP4_ADDR_TO_U32(&ipsettings.subnet, 255, 255, 255, 0);
   EOE_IP4_ADDR_TO_U32(&ipsettings.default_gateway, 192, 168, 9, 1);
   memcpy(ipsettings.mac.addr, mux_mac, sizeof(mux_mac));

zepandasky, 2021-09-16:
I've tried setting the mac address also. Didnt make a difference for me. I used it for both slaves with ethernet capabilities.
However i was able to use the ethercat as a passthrough for ethernet traffic, and receive an IP-adres on a pc connected to an ethercat slave ethernet port.

nakarlsson, 2021-09-16:
Ok seems, if I understand you correct, it works as switchport. 

You can try adding the entire EoE setip structure,  eoe_param_t. See if Wireshark accept that as a valid frame. I remember doing some fix that allowed for using parts of the eoe_param_t, but perhaps the slave doesn't accept that as a valid frame.
I find it odd that it doesn't respond at all, even if the struct is incomplete from its point of view.

nakarlsson, 2022-01-12:
@zepandasky  , any update? I've ordered a EL6601 but the delivery time is very long, sometime this sping. I ordered in November.

zepandasky, 2022-01-13:
Well my project got delayed quite a bit as well due to delivery issues as well. However I did succeed in making the EoE example work as a switchport. This was part of an R&D research, however it's extremely slow, to the point windows isn't even sure whether it  has a working internet connection sometimes. However I never got it working to assign an IP-adress to the module itself. 
What I do in my code is literally:

`
   size_of_rx = sizeof(rxbuf);
   wkc = ecx_EOEreadfragment(eoembx,
      &rxfragmentno,
      &rxframesize,
      &rxframeoffset,
      &rxframeno,
      &size_of_rx,
      rxbuf);

     if (wkc > 0)
       {
          if(slave == 17)
          {
             ecx_EOEsend(context, 6, 0, sizeof(rxbuf), rxbuf, EC_TIMEOUTRXM);
          }
          if(slave == 6)
             ecx_EOEsend(context, 17, 0, sizeof(rxbuf), rxbuf, EC_TIMEOUTRXM);
       }`

The above code is registered in the eoe_hook()
Please be ware it's pretty bad code but the whole goal was "is it possible to send TCP/IP traffic over the Ethercat protocol."

If slave 17 is activated it send the data to slave 6 and vica versa. Slave 6 is the slave which is connected to the LAN, slave 17 is connected to the end-device.

closed 2022-05-31 by zepandasky