OpenEtherCATsociety/SOEM #556: initCmds config | EL7342

jjrbfi, 2021-09-21:
Hello I'm trying to use a EL7342 but I'm looking forward to use the actual config I already have, because I don't get it moving:

<?xml version="1.0" encoding="ISO-8859-1"?>
<EtherCATMailbox><CoE><InitCmds>
	<InitCmd>
		<Transition>IP</Transition>
		<Timeout>0</Timeout>
		<Ccs>1</Ccs>
		<Index>61569</Index>
		<SubIndex>1</SubIndex>
		<Data>00001400</Data>
	</InitCmd>
</InitCmds></CoE></EtherCATMailbox>

Any hint?

Regards

ArthurKetels, 2021-09-21:
I am sorry but your question is much too vague to give a sensible answer.

ArthurKetels, 2021-09-22:
Reading information (tutorial.txt, this forum), experimenting (try to code something), and feedback (wireshark captures) are the name of the game here. You claim there are no examples, but there are, so clearly you missed something.

The XML code you have pasted above is nothing more than a check. Configuration is only allowed after checking that CoE object 0xf081 subindex 1 has value 0x00001400. And from reading the ENI file from the slave you can learn this is the version number of the slave. This makes perfect sense, check before you try to configure the slave whether the slave is actually the correct one.

But this slave has a ton of configuration registers (in the 0x8000-0x8fff range). You first have to set decent parameters in them before you could try to operate it. Then you have to define a suitable PDO mapping. Only then you can do ec_config() in SOEM. There are multiple examples here to see how to do that.

There is however no reason this slave would not work with SOEM, you just need to do your homework.

nakarlsson, 2022-01-12:
Closed due to inactivity

closed 2022-01-12 by nakarlsson