OpenEtherCATsociety/SOEM #567: Issues enabling STF06-EC with SOEM

mcoursolle, 2021-11-01:
Hi,
I am quite new to SOEM and CoE, but have successfully created code (Ubuntu 18.04) from the simple_test.c example to control HD actuators using a Copley (AEV) drive.
https://github.com/OpenEtherCATsociety/SOEM/blob/master/t…
Now I am trying to control a stepper (HT23-598D) from an Applied Motion (STF06-EC) drive. I am currently having 2x issues (not sure if they are related or not).

- When going from "Switch On"  to "Operation Enabled" (Writing 0x000F in Control Word), I do read that the drive is enabled (Status Word = 0x0237), and I hear a small sound at the stepper level. However, I can still easily move the stepper with my fingers. Using the STF Configurator tool and the serial interface, it feels quite locked and impossible to move once enabled.
 
- After the "Operation Enabled" mode is set, I am trying to set the operation mode to profile (Write 0x01 to 0x6060). I can read back from 0x6060 that 0x01 was set, but it never gets propagated to 0x6061 (mode of operation display).

Also, not sure if it is related (or why it was required), but I had to change the sleep at line 109 from simple_test.c from 5000 to 50000 for it to work. Otherwise, it would randomly go back to "Switched On Disabled".

Also, this is probably a beginner's question, but how should I use the XML file with SOEM?

Please find here some documentation on the used drive:
https://appliedmotion.s3.amazonaws.com/EtherCAT-User-Manual-920-0158-RevB.pdf
https://www.applied-motion.com/sites/default/files/STF_EtherCAT_XML_files_0.zip

Any help would be really appreciated.
Thank you!

Mathieu

mcoursolle, 2021-11-01:
[sdo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/7453095/sdo.txt
[map.txt](https://github.com/OpenEtherCATsociety/SOEM/files/7453096/map.txt)
)

mcoursolle, 2021-11-01:
[sdo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/7453100/sdo.txt)

mcoursolle, 2021-11-01:
[map.txt](https://github.com/OpenEtherCATsociety/SOEM/files/7453101/map.txt)

mcoursolle, 2021-11-01:
Please find above the output of slaveinfo -sdo and -map.

mcoursolle, 2021-11-01:
[wireshark-slaveinfo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/7453204/wireshark-slaveinfo.txt)

mcoursolle, 2021-11-01:
[wireshark-newcode.txt](https://github.com/OpenEtherCATsociety/SOEM/files/7453205/wireshark-newcode.txt)

mcoursolle, 2021-11-01:
Please find attached the output of wireshark with slaveinfo and from my new code (to be provided shortly).

mcoursolle, 2021-11-01:
[newcode.txt](https://github.com/OpenEtherCATsociety/SOEM/files/7453247/newcode.txt)

mcoursolle, 2021-11-01:
Please find attached the output of wireshark.

mcoursolle, 2021-11-01:
[wireshark-slaveinfo-sdo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/7454351/wireshark-slaveinfo-sdo.txt)

mcoursolle, 2021-11-01:
[wireshark-newcode.txt](https://github.com/OpenEtherCATsociety/SOEM/files/7454352/wireshark-newcode.txt)

mcoursolle, 2021-11-01:
[wireshark-pcapng.zip](https://github.com/OpenEtherCATsociety/SOEM/files/7454704/wireshark-pcapng.zip)

mcoursolle, 2021-11-01:
Just attached the pcapng format into a zip file, as Github doesn't seem to accept pcapng files?

ArthurKetels, 2021-11-01:
It looks like the slave needs synchronous PDO transfers. Either with SYNC0 or SM synchronous.  This means strict timing constraints on your master real time PDO thread. Your code does not do that, and probably the slave is waiting for stable timing until it goes to "operation enabled". Please read other post here on how to do that.

mcoursolle, 2021-11-02:
Ok thanks. I guess I'll have to read a bit more on the subject to fully understand that part. Does it have to do with the provided slave ESI file not being used? I know that for some master controllers that XML ESI file can simply be loaded. Is there a way to automatically configure SOEM from a XML ESI file?

ArthurKetels, 2021-11-02:
SOEM is a library to make your own master. So it is not a complete master by itself. If you want, you can make your own ESI parser and use that to build the configuration structs in SOEM. This ESI parser is currently not part of SOEM. The functions provided have an online method of configuring the slaves. This often works just out of the box but sometimes it needs some tweaking.

mcoursolle, 2021-11-03:
I continued reading and as a test, I added the code mentioned in the "PDO Assign and PDO Config" section of tutorial.txt.

```
uint16 map_1c12[4] = {0x0003, 0x1601, 0x1602, 0x1604};
uint16 map_1c13[3] = {0x0002, 0x1a01, 0x1a03};
retval += ec_SDOwrite(slave, 0x1c12, 0x00, TRUE, sizeof(map_1c12), &map_1c12, EC_TIMEOUTSAFE);
retval += ec_SDOwrite(slave, 0x1c13, 0x00, TRUE, sizeof(map_1c13), &map_1c13, EC_TIMEOUTSAFE);
```

It allowed the code to move forward and to be able to change the operation mode successfully (the stepper is now powered and locked), and 0x6061 reads the proper operation mode. However, the drive indicators says it is still in SAFE_OP state.

As a test I changed the code to set the default values from the ESI file:

```
uint16 map_1c12[5] = {0x0004, 0x1600, 0x1601, 0x1602, 0x1603};
uint16 map_1c13[5] = {0x0004, 0x1a00, 0x1a01, 0x1a02, 0x1a03};
retval += ec_SDOwrite(inID, 0x1c12, 0x00, TRUE, sizeof(map_1c12), &map_1c12, EC_TIMEOUTSAFE);
retval += ec_SDOwrite(inID, 0x1c13, 0x00, TRUE, sizeof(map_1c13), &map_1c13, EC_TIMEOUTSAFE);
```

That doesn't keep the drive in SAFE_OP, but the operation mode is not changed properly (0x6061 stays 0x00).

I am still trying to figure out how to use the SM. Would it be possible to get some hints as to what the difference is and/or where I could find some information on the proper settings please? Thank you!

mcoursolle, 2021-11-03:
Ok, I read a bit more on PDOs, and for some reason, it seems like my code fails to change the mode of operation (0x6060) when Control Word (0x6040) is mapped on 0x1C12. 

If I map something else (e.g. 0x60FF target velocity), then I can change the mode of operation, but still cannot move the stepper. The drive LED status indicates PRE_OP.

However, in the first case (mapped to 0x6040), I cannot change the mode of operation, but the drive LED indicates OPERATIONAL.

mcoursolle, 2021-11-03:
Ok, so current status is that I can enable to stepper motor, but am still trying to move it.

For some reason, as long as 0x6060 is not mapped in 0x1600 it seems to work, but not sure I understand why.

ArthurKetels, 2021-11-03:
I do not have access to your whole code, and what your are using seems to be a C++ wrapper around various SOEM functions. This makes it hard to exactly tell what your are doing in the code. It is best to only use SOEM library calls when you have questions on this site. 

Now for some guesswork. I assume your WriteData function internally calls ec_SDOwrite(). Then it is clear why your code does not work when you map 0x6040 in your PDO. At any moment an object is mapped in PDO you no longer can write it as SDO. Reading mostly works with both methods. It is preferable to do data manipulation in a slave with PDO as it is much more efficient than SDO read/write. So read the tutorial on how to access PDO data in SOEM.

mcoursolle, 2021-11-04:
"At any moment an object is mapped in PDO you no longer can write it as SDO"

Ok thanks! That is the part I didn't understand. Seems to work fine now!

mcoursolle, 2021-11-04:
One other issue I had however (which is resolved now), is that the drive would randomly go back to "Switched On Disabled" if I keep the 5000 delay at line 109 of simple_test.c. If I increase this delay to 50000 everything works fine. Is that delay defined somewhere in the drive ESI? How do I know which delay to set?

ArthurKetels, 2021-11-04:
From the slave datasheet  you can read that the minimum cycle time is 500us. So it is possible that your implementation / OS sometimes has a PDO cycle time shorter than this. The slave is then allowed to change state to "Switched on Disabled". You have to remember that a simple delay in your code does not have any real-time guarantees.

nakarlsson, 2022-01-12:
Closed due to inactivity

closed 2022-01-12 by nakarlsson