OpenEtherCATsociety/SOEM #568: Is SOEM Real-Time ?

msevnctkn, 2021-11-03:
hi everyone, i'm new in ethercat world and i am reading all things about etherCAT. I have a question about it. I'm using Windows OS and i want to control 6 servo motors over ethercat. Is it possible to control 6 servo motors over EtherCAT without any real time kernel (twincat, rtx64 etc.) using c++ thread or soem thread? 6 servo motors will rotate simultaneously. Is it possible. May be one of you say that what a bad question... or like this pls dont judge me as i said at starting of my page I'm very new in soem world.

ArthurKetels, 2021-11-03:
Hi, SOEM is a library you can use to build your own EtherCAT master. It has many targets you can build it for. Windows is one of them. But actually it is also the worst in terms of real-time performance.

Could build a system with SOEM using Windows that controls 6 servo drives? Yes, but the performance will be mediocre. And if your application will have any serious real-time requirements it will fail.

Decent performance can be had when using Linux with preempt-RT patch. For the best performance specialized real-time OS should be used or, even better, a bare-metal implementation. 

The overhead of SOEM is very low, and almost all of the timing jitter is caused by the OS and hardware it runs on.

I am sorry, but without more detailed descriptions of your requirements, it is not possible to be more specific.

msevnctkn, 2021-11-06:
Hi Arthur, thanks for the quick answer.

Could build a system with SOEM using Windows that controls 6 servo drives? Yes, but the performance will be **mediocre.** 
What do you mean by **mediocre**. I want to make PC-based CNC control software on the real-time OP.
My G-code command has 3 axes. Motors will rotate with respect to the x, y, and z axes simultaneously. Is it possible to make it without real-time OP?  

Decent performance can be had when using Linux with preempt-RT patch. For the best performance specialized real-time OS should be used or, even better, a bare-metal implementation.

Is preempt-RT patch easy to apply? I just looked and it looks a bit complicated.

ArthurKetels, 2021-11-06:
Good to tell me your application. At least I now know very roughly what you want to do. PC-based CNC control and EtherCAT is already solved as an open source project at [www.linuxcnc.org](url). Just download configure and use. Done.

Windows has latencies up to 100ms. For most CNC control this is too much. This assumes you have some feed-back loop implemented. When not you can just upload a list of coordinates and time-stamps to the drives. As long you keep the drive buffer alive the program will run without any hick-ups. This is how it is done with most windows based CNC control programs. This off course assumes drives with internal position feedback and interpolation capabilities.

The whole point of EtherCAT is that you can use it as a high speed bus with ultra low latencies. To be able to close the feed-back loop in your computer instead of the drive. And, as always, your system is as good as the weakest link. In this case your OS, windows.

On the other hand, I am a big advocate of trying things yourself. By all means set-up a windows system with EtherCAT and experiment. One way or the other you will learn al lot!

msevnctkn, 2021-11-13:
Hi, Arthur! After we chat, I ordered a Delta ASDA-A2. Could you write a couple of codes that turn the motor cyclically? I examined the example codes but, actually, I could not find the how to motor turns with respect to position data. Can you help me?

ArthurKetels, 2021-11-13:
Writing code for others is not exactly my preferred past-time. What is your contribution to this project so as to offset my efforts?

msevnctkn, 2021-11-22:
@ArthurKetels yeah Arthur, it was my bad. Just i wanted an example in detail with less codes. Yet not problem. Thanks a lot!

closed 2021-11-22 by ArthurKetels