OpenEtherCATsociety/SOEM #569: Rearranging the module in Wago 750-495

aditmak, 2021-11-08:
Hi,
 
I am new to EtherCat and I have a Wago 750 Bus Coupler. Attached to the coupler are 2 Digital Output modules and 1 3-Phase power management module (Wago 750-495). 

I ran the slaveinfo -map and got the mapping. I am successfully able to go to OP state and get inputs and outputs.
However, I have certain requirement in terms of IOMap. 

### Current 

```
Slave:1
 Name:WAGO Series 750 Bus Coupler
 Output size: 240bits
 Input size: 224bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000021 ID: 07500354 Rev: 00010002
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1400 L: 512 F:00010022 Type:2
 SM2 A:1800 L:  30 F:00010064 Type:3
 SM3 A:2400 L:  28 F:00010000 Type:4
 FMMU0 Ls:00000000 Ll:  30 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000001e Ll:  28 Lsb:0 Leb:7 Ps:2400 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 04
 CoE details: 33 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0xF200:0x01 0x01 BOOLEAN      K-Bus Cycle Overrun Flag Disable
  [0x0000.1] 0xF200:0x02 0x01 BOOLEAN      Input Process Data Hold Request
  [0x0000.2] 0xF200:0x03 0x01 BOOLEAN      Output Process Data Hold Request
  [0x0000.3] 0xF200:0x04 0x01 BOOLEAN      Output Process Data Clear Request
  [0x0000.4] 0x0000:0x00 0x0C
  [0x0002.0] 0xF200:0x05 0x10 UNSIGNED16   Diagnostics Control Word
  [0x0004.0] 0x7020:0x01 0x08 UNSIGNED8    Control
  [0x0005.0] 0x7020:0x02 0x08 UNSIGNED8    Byte 1
  [0x0006.0] 0x7020:0x03 0x08 UNSIGNED8    Byte 2
  [0x0007.0] 0x7020:0x04 0x08 UNSIGNED8    Byte 3
  [0x0008.0] 0x7020:0x05 0x08 UNSIGNED8    Byte 4
  [0x0009.0] 0x7020:0x06 0x08 UNSIGNED8    Byte 5
  [0x000A.0] 0x7020:0x07 0x08 UNSIGNED8    Byte 6
  [0x000B.0] 0x7020:0x08 0x08 UNSIGNED8    Byte 7
  [0x000C.0] 0x7020:0x09 0x08 UNSIGNED8    Byte 8
  [0x000D.0] 0x7020:0x0A 0x08 UNSIGNED8    Byte 9
  [0x000E.0] 0x7020:0x0B 0x08 UNSIGNED8    Byte 10
  [0x000F.0] 0x7020:0x0C 0x08 UNSIGNED8    Byte 11
  [0x0010.0] 0x7020:0x0D 0x08 UNSIGNED8    Byte 12
  [0x0011.0] 0x7020:0x0E 0x08 UNSIGNED8    Byte 13
  [0x0012.0] 0x7020:0x0F 0x08 UNSIGNED8    Byte 14
  [0x0013.0] 0x7020:0x10 0x08 UNSIGNED8    Byte 15
  [0x0014.0] 0x7020:0x11 0x08 UNSIGNED8    Byte 16
  [0x0015.0] 0x7020:0x12 0x08 UNSIGNED8    Byte 17
  [0x0016.0] 0x7020:0x13 0x08 UNSIGNED8    Byte 18
  [0x0017.0] 0x7020:0x14 0x08 UNSIGNED8    Byte 19
  [0x0018.0] 0x7020:0x15 0x08 UNSIGNED8    Byte 20
  [0x0019.0] 0x7020:0x16 0x08 UNSIGNED8    Byte 21
  [0x001A.0] 0x7020:0x17 0x08 UNSIGNED8    Byte 22
  [0x001B.0] 0x7020:0x18 0x08 UNSIGNED8    Byte 23
  [0x001C.0] 0x7000:0x01 0x01 BOOLEAN      Channel 1 Data
  [0x001C.1] 0x7000:0x02 0x01 BOOLEAN      Channel 2 Data
  [0x001C.2] 0x7010:0x01 0x01 BOOLEAN      Channel 1 Data
  [0x001C.3] 0x7010:0x02 0x01 BOOLEAN      Channel 2 Data
  [0x001C.4] 0x0000:0x00 0x0C
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x001E.0] 0xF100:0x01 0x01 BOOLEAN      K-Bus Cycle Overrun Flag
  [0x001E.1] 0xF100:0x02 0x01 BOOLEAN      Input Process Data Hold Ack.
  [0x001E.2] 0xF100:0x03 0x01 BOOLEAN      Output Process Data Hold Ack.
  [0x001E.3] 0xF100:0x04 0x01 BOOLEAN      Output Process Data Clear Ack.
  [0x001E.4] 0x0000:0x00 0x0B
  [0x001F.7] 0x10F3:0x04 0x01 BOOLEAN      New Message Available
  [0x0020.0] 0xF100:0x05 0x10 UNSIGNED16   Diagnostics Status Word
  [0x0022.0] 0x6020:0x01 0x08 UNSIGNED8    Status
  [0x0023.0] 0x6020:0x02 0x08 UNSIGNED8    Byte 1
  [0x0024.0] 0x6020:0x03 0x08 UNSIGNED8    Byte 2
  [0x0025.0] 0x6020:0x04 0x08 UNSIGNED8    Byte 3
  [0x0026.0] 0x6020:0x05 0x08 UNSIGNED8    Byte 4
  [0x0027.0] 0x6020:0x06 0x08 UNSIGNED8    Byte 5
  [0x0028.0] 0x6020:0x07 0x08 UNSIGNED8    Byte 6
  [0x0029.0] 0x6020:0x08 0x08 UNSIGNED8    Byte 7
  [0x002A.0] 0x6020:0x09 0x08 UNSIGNED8    Byte 8
  [0x002B.0] 0x6020:0x0A 0x08 UNSIGNED8    Byte 9
  [0x002C.0] 0x6020:0x0B 0x08 UNSIGNED8    Byte 10
  [0x002D.0] 0x6020:0x0C 0x08 UNSIGNED8    Byte 11
  [0x002E.0] 0x6020:0x0D 0x08 UNSIGNED8    Byte 12
  [0x002F.0] 0x6020:0x0E 0x08 UNSIGNED8    Byte 13
  [0x0030.0] 0x6020:0x0F 0x08 UNSIGNED8    Byte 14
  [0x0031.0] 0x6020:0x10 0x08 UNSIGNED8    Byte 15
  [0x0032.0] 0x6020:0x11 0x08 UNSIGNED8    Byte 16
  [0x0033.0] 0x6020:0x12 0x08 UNSIGNED8    Byte 17
  [0x0034.0] 0x6020:0x13 0x08 UNSIGNED8    Byte 18
  [0x0035.0] 0x6020:0x14 0x08 UNSIGNED8    Byte 19
  [0x0036.0] 0x6020:0x15 0x08 UNSIGNED8    Byte 20
  [0x0037.0] 0x6020:0x16 0x08 UNSIGNED8    Byte 21
  [0x0038.0] 0x6020:0x17 0x08 UNSIGNED8    Byte 22
  [0x0039.0] 0x6020:0x18 0x08 UNSIGNED8    Byte 23
End slaveinfo, close socket
End program
```

### Required 

I want to have IOMap similar to the way my modules are arranged in wago  (0x7000, then 0x7010, ...0x7fff). (similar for Inputs)
The main goal of doing this is to be certain and create a predefined struct for the entire coupler so that if new hardware is introduced, I only need to extend my struct and not worry about the mapping modifying.

```
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0xF200:0x01 0x01 BOOLEAN      K-Bus Cycle Overrun Flag Disable
  [0x0000.1] 0xF200:0x02 0x01 BOOLEAN      Input Process Data Hold Request
  [0x0000.2] 0xF200:0x03 0x01 BOOLEAN      Output Process Data Hold Request
  [0x0000.3] 0xF200:0x04 0x01 BOOLEAN      Output Process Data Clear Request
  [0x0000.4] 0x0000:0x00 0x0C
  [0x0002.0] 0xF200:0x05 0x10 UNSIGNED16   Diagnostics Control Word
  Changed : [0x0004.0] 0x7000:0x01 0x01 BOOLEAN      Channel 1 Data
  Changed : [0x0004.1] 0x7000:0x02 0x01 BOOLEAN      Channel 2 Data
  Changed : [0x0004.2] 0x7010:0x01 0x01 BOOLEAN      Channel 1 Data
  Changed : [0x0004.3] 0x7010:0x02 0x01 BOOLEAN      Channel 2 Data
  Changed : [0x0004.4] 0x0000:0x00 0x0C
  [0x0005.0] 0x7020:0x01 0x08 UNSIGNED8    Control
  [0x0006.0] 0x7020:0x02 0x08 UNSIGNED8    Byte 1
  [0x0007.0] 0x7020:0x03 0x08 UNSIGNED8    Byte 2
  [0x0008.0] 0x7020:0x04 0x08 UNSIGNED8    Byte 3
  [0x0009.0] 0x7020:0x05 0x08 UNSIGNED8    Byte 4
  [0x000A.0] 0x7020:0x06 0x08 UNSIGNED8    Byte 5
  [0x000B.0] 0x7020:0x07 0x08 UNSIGNED8    Byte 6
  [0x000C.0] 0x7020:0x08 0x08 UNSIGNED8    Byte 7
  [0x000D.0] 0x7020:0x09 0x08 UNSIGNED8    Byte 8
  [0x000E.0] 0x7020:0x0A 0x08 UNSIGNED8    Byte 9
  [0x000F.0] 0x7020:0x0B 0x08 UNSIGNED8    Byte 10
  [0x0010.0] 0x7020:0x0C 0x08 UNSIGNED8    Byte 11
  [0x0011.0] 0x7020:0x0D 0x08 UNSIGNED8    Byte 12
  [0x0012.0] 0x7020:0x0E 0x08 UNSIGNED8    Byte 13
  [0x0013.0] 0x7020:0x0F 0x08 UNSIGNED8    Byte 14
  [0x0014.0] 0x7020:0x10 0x08 UNSIGNED8    Byte 15
  [0x0015.0] 0x7020:0x11 0x08 UNSIGNED8    Byte 16
  [0x0016.0] 0x7020:0x12 0x08 UNSIGNED8    Byte 17
  [0x0017.0] 0x7020:0x13 0x08 UNSIGNED8    Byte 18
  [0x0018.0] 0x7020:0x14 0x08 UNSIGNED8    Byte 19
  [0x0019.0] 0x7020:0x15 0x08 UNSIGNED8    Byte 20
  [0x001A.0] 0x7020:0x16 0x08 UNSIGNED8    Byte 21
  [0x001B.0] 0x7020:0x17 0x08 UNSIGNED8    Byte 22
  [0x001C.0] 0x7020:0x18 0x08 UNSIGNED8    Byte 23

```

I have attached the ESI file. [WAGO_750_354_21.zip](https://github.com/OpenEtherCATsociety/SOEM/files/7497177/WAGO_750_354_21.zip)

Please let me know if there is the way to do the required mapping.

ArthurKetels, 2021-11-08:
Well, bad luck for you. The slave is a bridge type that converts K-bus modules to EtherCAT. As stated in the ESI file it only supports PdoUpload and not PdoConfig or PdoAssign. This means that the mapping of IO data of modules to EtherCAT is determined at runtime by the slave and the master can not change that. 

You could read out the module configuration and figure out the mapping. You then have to make a translation table from your desired mapping to the actual mapping.

The easiest way is to use only real EtherCAT slaves and no K-bus modules. But then you have to switch from Wago to Beckhoff hardware.

And all of this has nothing to do with SOEM.

aditmak, 2021-11-10:
Thank you Arthur for explaination. Just 1 more question. 

Is there a way which gives me the IOmap address from index and subindex? 
I have seen how it is implemented in slaveinfo.c

nakarlsson, 2022-01-12:
Sorry, no relationship of SDO vs IOmap exist. It is setup once

closed 2022-01-12 by nakarlsson