OpenEtherCATsociety/SOEM #570: link error when using SOEM as a dependency in another project

manumerous, 2021-11-08:
When I tried to use SOEM as a dependency in another project (using catkin build from [ROS](https://www.ros.org/) ) I got the following error message:

```
/usr/bin/ld: /usr/local/lib/libsoem.a(ethercatcoe.c.o): relocation R_X86_64_PC32 against symbol `ecx_context' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
```

The issue could me fixed by adding the following line to the CMakeLists.txt of SOEM:
`set(CMAKE_POSITION_INDEPENDENT_CODE ON)`

And going again through:

```
cd build/
cmake ..
make install
```

Is there a certain reason the position independent code flag is not set for this library? If not I would be happy to add it as a PR to the repository in order to make using the lib easier for others. :)

Best regards, 
Manuel

ArthurKetels, 2021-11-08:
Using catkin is horrible, but to everyone his own poison I guess. What happens is that you link a dynamic library against a static library (SOEM). As you noticed that does not work but can be overcome by generating the static lib as PIC.  If you want to, you can do this. But why not generate a dynamic lib?

As a better solution for real-time code I would prefer to link all libs static. Less overhead. But that kind of implies not using ROS.

manumerous, 2021-11-08:
Thanks for your answer Arthur! Unfortunately, the interfaces of my project are given so I can not link everything in a static way. I will stick to this fix then.

closed 2021-11-08 by manumerous