OpenEtherCATsociety/SOEM #571: EtherCAT ID Issues

mcoursolle, 2021-11-10:
I currently have 2x daisy-chained drives (Copley AEV and Applied-Motion STF-EC) connected and controlled from SOEM.

I manually set the S2 switch on both of them so that the AEV is 1, and STF-EC is 2.

However, after the call to ec_config_init(FALSE), they seem to be in the opposite index. I tried to change the switch position on the drives to change their ID, but it doesn't seem to make any difference.

AEV - S2 = 1
STF - S2 = 2

Expected:
ec_slave[1].name = "AEV"
ec_slave[2].name = "STF-EC"

However, I always get the following, even if I change the switch positions.
ec_slave[1].name = "STF-EC"
ec_slave[2].name = "AEV"

Am I missing something to tell SOEM to use the ID from the hardware switch?

Thank you!

mcoursolle, 2021-11-10:
Ok, seems like aliasadr is what I was looking for.

I am right that ec_SDOwrite(...) and ec_SDOread(...) are however expecting the idx from ec_slave, and not the aliasadr as an input.

I guess I could create a map between the idx and the aliasadr to know which idx to use when calling ec_SDO<read/write>(...), but is there an easier SOEM functionality to do so?

Thank you!

ArthurKetels, 2021-11-10:
The default slave discovery algorithm takes the slaves in natural port order. If you want you can code a customized function that looks at the aliasadr to program the configured address. But what do you do when a slave has no aliasadr, or two slaves with the same?

The ec_slave array follows the configured address order. So yes, ec_SDOwrite() needs the index as defined in ec_slave.

I guess the easiest way is to map between custom order and configured slave order.

mcoursolle, 2021-11-10:
What would be considered the natural port order?

I am currently connected in the following order:

Computer (SOEM) -> AEV (ID 1) -> STF (ID 2)

SOEM then stores the idx as such:
ec_slave[1] -> STF
ec_slave[2] -> AEV

So it would be from the last one in the chain down to the 1st one? Not sure why, but I was expecting the opposite.

Thank you!

ArthurKetels, 2021-11-10:
Then most likely you have swapped the in and out port of your AEV slave. The order should always be master->slave1in->slave1out->slave2in.

mcoursolle, 2021-11-10:
Ok, I feel bad for asking now...just realized I indeed inverted In and Out on the AEV drive.
I guess I wasn't expecting the code to work at all in that configuration.
Thank you!

ArthurKetels, 2021-11-12:
Problem solved, issue closed.

closed 2021-11-12 by ArthurKetels