OpenEtherCATsociety/SOEM #572: Problems implementing a basic version of ADS Over EtherCat (AoE)

eliapellegrino, 2021-11-10:
Dear all,
I am currently trying to work with the EL6224, an IO-Link Master EtherCat terminal from Beckhoff. In order to fully use it, I need to implement a basic version of AoE, to exchange configuration data. I read the corresponding documentation on this, in particular ETG.5001.6220 and ETG.1020.
I manage to use the AoE with TwinCat and I took some capture with Wireshark. The initialization phase is pretty straightforward (it is just a FPWR command with some standardized data to send), but I am already stuck in here.
I am currently working on this machine:
```
Linux 4.19.89-xeno-ipipe-3.1 #1 SMP Tue Sep 29 16:46:56 CEST 2020 x86_64 x86_64 x86_64 GNU/Linux
```
using [RTNet](http://www.rtnet.org/).

In particular what I got from TwinCat is:
[image]
where the AMS request is sent 2 times, one in INIT->PREOP and the other in PREOP->SAFEOP, and the response arrives few packets later.

In my current implementation, what I obtain is this:
[image]
where the request is not accepted by slave (you can see it from the wkc of the second packet that is equal to 0).
To achieve this result, I had simply copied the packet to send (with the right modification according to ETG.1020), and sent it using "FPWR" command (as in TwinCat).
What I would like to know is:
1) Why the AoE initialization is performed twice?
2) Could the second packet exchanged in the request be mandatory in the the communication?
3) Is this the right place to ask, or should I directly contact Beckhoff to get into the details of slave's firmware?

Thank you very much for the attention.

Additional information:
[Wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/7513987/Wireshark.zip)
[EL6224 Esi file](https://github.com/OpenEtherCATsociety/SOEM/files/7514002/EL6224.zip)

ArthurKetels, 2021-11-10:
You are making things too complicated. The function that you need in SOEM is ecx_mbxsend(). It will send a mailbox buffer using the proper protocol and auto resend and recovery. The other half is ecx_mbxreceive().

The next thing that is going wrong is that you try to send a mailbox when the slave is still in INIT. You first have to put the slave in pre-OP in order for the mailbox buffers to become active.

To answer your questions:
1. TwinCAT does not do AoE init twice. Each time other data is send.
2. AoE is an encapsulated protocol. You need to dig deeper to see what data is actually transferred to the slave.
3. No not really the right place for asking slave details. But the implementation of AoE in SOEM is definitly a proper question. 

I have never had the need to implement AoE in SOEM and because I have no slave to test it on the urge to implement it is rather low.

eliapellegrino, 2021-11-11:
Currently I am already using ``` ecx_mbxsend() ``` to send the "FPWR" command: what I have done basically, it is to copy the same function of ``` ec_SDOwrite()```, due to the fact that the higher protocol (Mailbox) is the same (of course changing the mailbox header). I don't know if you open the wireshark capture, but what I have obtained is this:
[image]
that it is set according to the [ETG.1020 (Sec 7.4, AoE Initialization)](https://www.ethercat.org/login.aspx?ReturnUrl=%2fmemberarea%2fdownload%2fETG1020_V1i2i0_S_R_ProtocolEnhancements.zip).
TwinCat sends this:
[image].
The only difference in the 2 packets are:
1) The master netId, that TwinCat initializes at 10.240.23.17.3.1, while my master netId is set to 00.00.10.0.0.1 (that is the IP address assigned to my NIC though RTNet, in the config file (but I don't know really if it must be a predefined number or not).
2) The configaddr of the slave to configure, that in TwinCat is set to 0x03ea (1002) while in SOEM is set to 0x1002 (4098)
3) The data append after the 6 bytes of ADS data, that in my implementation is zero-padding (as SOEM performs), while in TwinCat is another "FPWR" command
[image]
as you can see above.
Of course, I can't know what my slave expects from AoE master (nor in ESI file neither in datasheet is written), but currently I think that I am doing the right thing.

Any thought about this?

ArthurKetels, 2021-11-11:
You forgot this thing 

> You first have to put the slave in pre-OP in order for the mailbox buffers to become active.

eliapellegrino, 2021-11-11:
Sorry, but what do you mean by "become active?"

ArthurKetels, 2021-11-11:
A SyncManager is a controller that takes care of data consistency of a dual-ported memory section. One side is for the EtherCAT bus the other for the slave controller. SOEM configures a SM and enables it for the EtherCAT bus. The slave side is not activated until the slave firmware enables it. This enabling is done when the master (SOEM) request the state change from init to pre-op. The slave first checks if the SM configuration is acceptable and only then enables the slave side.

The effect of a SM that is not enabled on the slave side is that it will return a workcounter of 0. The master can thus check if the SM is active. For mailbox transfer there is another constraint. When a mailbox is full and not read by the other side, all new attempts to write to the mailbox will fail with a workcounter of 0. This happens when the master is faster with writing than the slave can read.

eliapellegrino, 2021-11-12:
> You forgot this thing
> 
> > You first have to put the slave in pre-OP in order for the mailbox buffers to become active.

This totally solves my problem, thank you!

closed 2021-11-12 by eliapellegrino