OpenEtherCATsociety/SOEM #573: slaveinfo example run into issue

craigtao, 2021-11-15:
If I don't config NIC with IP address,
When the program executes here, the bind function returns -1.

int ecx_setupnic(ecx_portt *port, const char *ifname, int secondary)
{
    ……
    r = bind(*psock, (struct sockaddr *)&sll, sizeof(sll));
    ……
}

If I configure the network card IP address, the bind function executes correctly.

ethercat network need IP address?

ArthurKetels, 2021-11-15:
It used to be no issue at all to use SOEM on a NIC that had no IP address. But in newer kernels there is some power saving logic that takes the NIC down when it does not see any "regular activity". I am not sure what the exact logic is.

Until now I have found two solutions:
a) Make sure a slave is connected, then bring the NIC down and up again with the ifconfig tool, then start SOEM.
b) Assign a fixed IP address to the NIC.

craigtao, 2021-11-16:
> It used to be no issue at all to use SOEM on a NIC that had no IP address. But in newer kernels there is some power saving logic that takes the NIC down when it does not see any "regular activity". [...]

closed 2021-11-16 by craigtao