OpenEtherCATsociety/SOEM #574: soem with multiple drives

k-jayanth, 2021-11-17:
Hi everyone,
i am trying to control my 2 actuator simultaneously using SOEM, These are some issues i am facing

1. i was able to map the objects correctly and get into operation state but after getting into operation state when i was setting the target position sometimes it is updating the value correctly and sometimes it changes for 1 drive  for only and sometimes is not updating for any drive. why this is not consistent?
2. Sometimes position demand value gets 0 after motor starts operation and motor stops suddenly. i don't know if its a communication loss problem or something else
3. how can i determine the optimum time for giving sleep command after every cycle
4.   #487 i have read this but is there any example for how can i achieve in soem or what what things i have to modify (in soem) or take care when multiple slaves are connected.

please suggest any other issues if i am missing or doing it wrong.
Attaching the Wireshark capture any help is appreciated
[wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/7551916/wireshark.zip)

ArthurKetels, 2021-11-17:
Well, you do not give me a lot of information to work with. If you don't understand #487 then do not bother to run servo slaves in a stable way. I can see from the wireshark traces your code does not really implement any of the important things mentioned there. Give your slaves some time to synchronize on the PDO transfers before changing the EtherCAT state. And do not fill the bus just to check if slaves have transitioned state. 

To answer your questions (at least a bit):
1. Because your code is not consistent.
2. It is your code. Your PDO transfers are not stable.
3. Have a look at red_test code. And also read what others have posted here. There is a wealth of information.

SOEM works perfectly with many slaves (my highest count is 1600 at the same time). You are the architect, SOEM is just providing the bricks and tools.

k-jayanth, 2021-11-22:
Thank you @ArthurKetels for the fast response

i have followed your suggestions and made some changes to my code and also read other posts about synchronization, found this post very helpfull
https://github.com/OpenEtherCATsociety/SOEM/issues/520#issuecomment-852381531

These are the steps i am following

1. ec_init interface
2. ec_config_init     // INIT to PRE-OP state
3. if any fault reset by control word individually
4. do mapping for slaves individually
5. sleep 500ms
5. ecx_context.manualstatechange = 1;
6. ec_config_map
7. ec_configdc
8. change state to EC_STATE_SAFE_OP
9. ec_dcsync0(this->slave_number, TRUE, 1000000U * 5, 0);
10. sleep 50ms
11. change state to EC_STATE_OPERATIONAL
12. do operation like changing position
13.  change state to EC_STATE_INIT
14. ec_close

Now i am able to control the drives consistently, but not sure about the sleep time(how much i should give?),  state transitions timings.
please suggest how i can improve it further or am i missing something still to make it real time

Attaching the Wireshark capture with the ESI file
[wireshark with ESI.zip](https://github.com/OpenEtherCATsociety/SOEM/files/7580746/wireshark.with.ESI.zip)

ArthurKetels, 2021-11-22:
Well, the Wireshark trace certainly looks better. But there are still timing issues with your PDO. It would be nice if you could post your code. At least then I can give a more informed response.

k-jayanth, 2021-11-23:
Thank you @ArthurKetels for reviewing.
Here is the code 

[sample.zip](https://github.com/OpenEtherCATsociety/SOEM/files/7587788/sample.zip)

k-jayanth, 2021-11-26:
Yes, i have seen it and using the ecatthread thread. what else can i use to match the PDO cycle

Can you say which events i have to monitor in wireshark log for analysing the PDO cycles (for any timing issues or any mismatches)

mage9353, 2021-12-02:
Hi, k-jayanth 
Please, explain why serialization of events is not good for servo drives?
I have to make EhterCAT master to control servo drive without an OS and I am planning to use just a timer interrupt

k-jayanth, 2021-12-06:
Hi, @mage9353 

i think serialization is ok but the PDO cycles has to be in cyclic loop for fixed timing
@ArthurKetels  can give you more guidance in this, and please read other useful posts

k-jayanth, 2021-12-06:
Hi, @ArthurKetels  
could you send your mail or contact, so that we can discuss to solve this synchronization problem that i am facing.

ArthurKetels, 2021-12-06:
I simply meant that PDO transfers are asynchronous from configuration code. Have a look at red_test.c. Also read the many posts about this subject here. (sometimes I feel like a parrot, repeating the same thing over and over).

nakarlsson, 2021-12-07:
Yes, we need to improve the documentation and sample code to ease the burden.

nakarlsson, 2022-01-12:
Closed due to inactivity

closed 2022-01-12 by nakarlsson