OpenEtherCATsociety/SOEM #575: SOEM with DC and cyclic counter

wengertom, 2021-11-24:
Hello together,

I have a problem with  my EtherCAT master using SOEM. For controlling my robot over EtherCAT I'm using a ROS2 wrapper around SOEM package.
My robot EtherCAT slave has DC and has a cyclic counter that must be incremented by one every cycle (From the Value 0 to 3). With this method the slave is sure not to miss data packages. 
I reached the point where I get to operational state with DC activated and a DC timer of 1 ms.  My cyclic loop is sending every run the new status word with the cyclic counter. 

Everything seems alright and works fine for some time (sometimes only a few sec and other times it runs for a couple min). And the it suddenly stops. I'm using preempt_rt patch on my laptop and I traced latency. On the system under full load my max latency was never above 60 microseconds. 

After receiving the above issue I started tracing my cyclic loop. I realized that besides the know latency to my DC timer of around 60 microseconds, I rarely have spikes up to 3 ms that break my real time loop. If I check my code without sending and reading PDO data with "ec_send/receive_processdata" I never get those latency spikes. 
So could it be that the call from my real time loop is getting blocked by another system internal thread with even higher prio?

Do you guys know if there are some interrupts from the system with high priority that could potential block the NIC for my loop? 

I am thankful for any help or information. Probably someone else came across a similar issue.

Cheers 
Tom

ArthurKetels, 2021-11-24:
Most likely a problem with  your network socket. You can change the NIC behavior with ethtool. Please read : https://github.com/OpenEtherCATsociety/SOEM/blob/master/drvcomment.txt

wengertom, 2021-12-01:
Hi @ArthurKetels,
thanks a lot for your reply. I was thinking if there are some battery saving options for the NIC as well. I will definitely have a look at the ethtool and try it out. 

I close this issue for now since this could do the trick and if not I need to further investigate and also check the EtherCAT connection with wireshark.

Thank you!
Cheers

closed 2021-12-01 by wengertom