OpenEtherCATsociety/SOEM #576: QNX platform support

briansoe66, 2021-12-02:
Hello, are there plans for QNX platform support?

I see that Opal-RT has a QNX fork: https://github.com/Opal-RT-Technologies/SOEM. But the author is no longer working at Opal-RT, and the licensing of that fork is unclear.

ArthurKetels, 2021-12-02:
QNX is not exactly free software. So if someone pays me for it, why not? The implementation of SOEM to any OS is not very complicated. It is even easy to do yourself. And when you look at the Opal-RT fork the changes in nicdrv and osal and oshw are quite easy. The question if that implementation is most optimal for QNX I can not answer.

briansoe66, 2021-12-02:
I see, thank you for the response!

nakarlsson, 2021-12-17:
@briansoe66  , can this be closed?