OpenEtherCATsociety/SOEM #577: Change the state of a single slave

Davidinos, 2021-12-13:
Goodmorning,
I'm using SOEM succefully and I can send SDO and using PDOs to communicate with an ethercat motor drive.
Currently I'm writing a function to change one slave state back and forth among OPERATIONAL, SAFE_OP, PRE_OP and INIT; keeping, in the meantime, the other slaves in OPERATIONAL state.
I've got some questions:

1. Is it necessary to create a group for every single slave (using ec_config_map_group()) so that I can stop PDO communication of a single slave, while keeping the others alive?
2. The function would be something like the following:

`	
void ecat_changeState(int slaveNum, uint16 state)
{
       ec_readstate();
        if(ec_slave[slaveNum].state != state)
	{
		if((ec_slave[slaveNum].state == EC_STATE_INIT) |
			(ec_slave[slaveNum].state == EC_STATE_PRE_OP) |
			(ec_slave[slaveNum].state == EC_STATE_SAFE_OP) |
			(ec_slave[slaveNum].state == EC_STATE_OPERATIONAL))
		{
			ec_slave[slaveNum].state = state;
			if(state == EC_STATE_OPERATIONAL)
			{
				ec_send_processdata();
				ec_receive_processdata(EC_TIMEOUTRET);
			}

			if (ec_statecheck(slaveNum, state, EC_TIMEOUTSTATE * 4) != state)
			{
				ec_readstate();
				PRINTF("Slave %d State=0x%2.2x StatusCode=0x%4.4x : %s\n",
					 ec_slave[slaveNum].state, ec_slave[slaveNum].ALstatuscode, ec_ALstatuscode2string(ec_slave[slaveNum].ALstatuscode));
			}
		}
	}
}` 
Do you think it might work?

Thank you.

nakarlsson, 2021-12-13:
1. No
2.  Check simple_test "ecatcheck", it'll recover a lost slave. In some sense the individual handling of a single slave can be handled in the same way.

Davidinos, 2021-12-16:
I think I solved the issue, thank you for your answer.
I close the case

closed 2021-12-16 by Davidinos