OpenEtherCATsociety/SOEM #578: BLDC Motor control

bhargavah, 2021-12-21:
Hi Guys,

This is not an issue, instead of a question. I haven't seen any questions on this topic here.

My idea is to have the EtherCAT based stack for controlling the BLDC motor. I plan to have 1 master & 1 slave and the BLDC motor is connected to the slave. At master, I would configure SOEM and at the slave, SOES. This way, the EtherCat stack/network would be created between the master and the slave which should be ok to control the BLDC motor.

In a nutshell.

Master(SOEM) -> Slave(SOES) -> BLDC Motor.

For explanation, we can think of Master as our laptop and slave as a Raspberry/Arduino/Any other board(might be attached to the driver separately) and BLDC motor would be connected with the slave.

I am looking for suggestions here to know if there is any technical deficiency in this architecture. I would be happy to share more details if needed.

Thank a lot in advance.

ArthurKetels, 2021-12-22:
What you propose is a very common solution. There are already many products for sale that work exactly like this.

bhargavah, 2021-12-22:
@ArthurKetels Thanks a lot for your reply.

I see. It would be excellent if you can comment on this architecture. 
Master(SOEM) -> Slave(SOES) -> Shield -> BLDC Motor

The shield is nothing but a board between Slave board & motor.

Also, it is good to know that such products are already there. It would be highly appreciable if you can point me out some links so that I can go ahead to try them.

nakarlsson, 2022-01-12:
What exactly is missing? You need of the shelve hardware? Evaluation boards?

bhargavah, 2022-01-12:
@nakarlsson Hi there, I did not get what you just said here.

nakarlsson, 2022-01-12:
>It would be highly appreciable if you can point me out some links so that I can go ahead to try them.

Links to what? COTS HW that is ready to use? Or COST HW on which you can develop a SOEM -> SOES -> motor setup?

dfreiberger, 2022-02-20:
Example COTS BLDC driver (combine with EK1100): https://www.beckhoff.com/en-us/products/i-o/ethercat-terminals/el7xxx-compact-drive-technology/el7211.html

Example DIY EtherCAT slave shield: https://www.bausano.net/en/hardware/ethercat-e-arduino/ethercat-and-raspberry.html

Or http://www.simplerobot.net/2018/06/rpi-ethercat-master-picat.html

nakarlsson, 2022-03-08:
@bhargavah , examples given so we’ll close the issue

closed 2022-03-08 by nakarlsson