OpenEtherCATsociety/SOEM #579: raspberry pi 4 support soem master?

craigtao, 2021-12-27:
I have question, 
thanks

Craig

ArthurKetels, 2021-12-27:
First hit on google : https://github.com/thanhtam-h/soem-w5500-rpi

craigtao, 2021-12-28:
> 

thanks.
I hope it's not in this way, Wiznet W5500 Ethernet chip with SPI interface

The Raspberry Pi 4 uses a separate BCM54213PE PHY, and the SoCs are connected using a dedicated RGMII link.
3B+ Gigabit ABI USB 2.0 Hub way .

https://www.broadcom.com/products/ethernet-connectivity/phy-and-poe/copper/gigabit/bcm54213pe

BCM54213PE
Contact Sales
Check Inventory
Request Info
Single Port RGMII Gigabit Ethernet Transceiver
Overview
Specifications
The Broadcom® BCM54213PE is a triple-speed 1000BASE-T/100BASE-TX/10BASE-T Gigabit Ethernet (GbE) transceiver integrated into a single monolithic CMOS chip. The device performs all physical-layer functions for 1000BASE-T, 100BASE-TX, and 10BASE-T Ethernet on standard category 5 UTP cable. 10BASE-T can also run on standard category 3, 4, and 5 UTP. The BCM54213PE is a highly integrated solution combining digital adaptive equalizers, ADCs, phaselocked loops, line drivers, encoders, decoders, echo cancellers, crosstalk cancellers, and all required support circuitry. Based on Broadcom’s proven digital signal processor technology, the BCM54213PE is designed to be fully compliant with RGMII, allowing compatibility with industry-standard Ethernet MACs and switch controllers. Designed for reliable operation over worst-case category 5 cable, the BCM54213PE automatically negotiates with its link partner to determine the highest possible operating speed. The device detects and corrects most common wiring problems. The BCM54213PE features CableChecker™ diagnostics, which detects common cable problems including shorts, opens, and cable length.
 
Features
• Single-chip integrated triple-speed Ethernet transceiver
• 1000BASE-T IEEE 802.3ab
• 100BASE-TX IEEE 802.3u
• 10BASE-T IEEE 802.3
• 100BASE-EFX
• Supports RGMII MAC interface
• Supports SOFT-RESET
• Ethernet@Wirespeed™
• Integrated voltage regulator
• Trace matched output impedance
• Line-side loopback
• Low EMI emissions
• Cable plant diagnostics
• Robust CESD tolerance
• Support for jumbo packets up to 10 KB
• Detection and correction of pair swaps
• (MDI crossover), pair skew, and pair polarity
• Advanced power management
• Super-isolate mode
• Wake-on-LAN (WOL)
• IEEE 802.3az-compliant support
• Support for native EEE MACs
• Support for legacy non-EEE MAC using AutogrEEEn® mode
• AutogrEEEn+
• 36-pin QFN package

nakarlsson, 2021-12-28:
It is an answer to your question? SOEM can run on RPi and almost any HW with an MAC/PHY.

nakarlsson, 2022-01-12:
You can run SOEM on RPi.

closed 2022-01-12 by nakarlsson