OpenEtherCATsociety/SOEM #580: IP map configuration

SakshamAnzo, 2021-12-28:
How to determine the size of the IOmap in SOEM?

nakarlsson, 2021-12-28:
The return value of the IO mapping function is the size of the IOmap. ecx_config_map and similar.

SakshamAnzo, 2021-12-28:
Yes I understand but we pass the pointer to it first that is pointing to a block of memory sometimes 4096 or 128.?

How to determine this?

nakarlsson, 2021-12-28:
It is up to you, how much payload will you allow SOEM to configure? If you allow only a single EL1008, a single byte will do.
Allowing 2xEL1008 will need two byte…

So, setup a the IOmap to maximum allowed payload plus som extra. Of course you should do a check on startup that you don’t exceed the available IOmap.

SakshamAnzo, 2021-12-28:
Okay Understood 

Thank you for following up

closed 2021-12-28 by SakshamAnzo