OpenEtherCATsociety/SOEM #583: Can't we control the PDOs through the SDO map if the exist for the same object?

SakshamAnzo, 2022-01-06:
I hope you guys are doing great.

I come to know about you through github SOEM query and i am here to consult a query about my servo drive . I wanted to drive a Ruking servo using ethercat . For now i understand that there will be PDOs and SDOs that will help in setting up parameters for the servo drive and thus move the rotor.The PDOs are cyclic in application and SDOs are only processed in some immediate manipulation of a parameter. 

My query is i am passing a default PDO that is Control Word with address 6040h (That is what i have received in the slaveinfo -map) now if i check (slaveinfo -sdo) it is also giving me the parameter Control Word with same logical address(6040h).

Now as it is mapped as an SDO as well I though i could go with an easy way and use ecx_SDOwrite to manipulate the control word and change the machine state.

but my wireshark capture is giving me SDO about code that is "0601 0000h" which depicts unsupported access to an object.

I am stuck at this point and not able to proceed further.

if you could please make some time in routine and provide few samples to run the motor or anything you find relevant .I would be able to learn it and being grateful for your support:)

nakarlsson, 2022-01-06:
You have 2 issues Open on the same topic? More for us to administrate :(.

I suggest you check old Issues on the topic.
You can’t control PDO mapped SDOs via SDO read/write. The way to do it is to read/write data via the IOmap as mentioned in several issues and samples.

nakarlsson, 2022-01-12:
No real issue, use the IOmap to control the slaves

closed 2022-01-12 by nakarlsson