OpenEtherCATsociety/SOEM #584: Writing data to EL2004 Beckhoff 4 channel digital output

davidDS96, 2022-01-06:
Dear community,

As the title of my issue already states, we want to write data to a EL2004 4 channel digital output terminal (4 digital outputs).

At this moment, we are able to configure the slave correctly and get it into operational state, however when we want to set one channel high, nothing seems to happen (the light on).
Currently, we are using following snippet of code:
``` c++
void set_output_bit (uint16 slave_no, uint8 module_index, uint8 value)
{
   /* Get the the startbit position in slaves IO byte */
   uint8_t startbit = ec_slave[slave_no].Ostartbit;
//    std::cout << startbit << std::endl;
   /* Set or Clear bit */
   if (value == 0)
      *ec_slave[slave_no].outputs &= ~(1 << (module_index - 1 + startbit));
   else
      *ec_slave[slave_no].outputs |= (1 << (module_index - 1 + startbit));
}
```
We are not sure if this is the right thing to do ...
When we read the value of the channel's bytes using ```*ec_slave[3].outputs``` it changes, but the control lights on the terminal doesn't go on or off.

Can someone provide us with the correct information of setting bits on a digital output terminal?

Thanks in advance!

David

nakarlsson, 2022-01-06:
Do you actually send the process data? Is it realy in OP? The run LED got a steady green light?

If you keep it simple, setting the entire byte 0xFF. Does the moduls det all ”On”?

post a Wireshark in start-up and when running in OP.

davidDS96, 2022-01-07:
The run LED was "On", but we were sending the data only once.

We have followed the example file ```simple_test.c``` and now we are able to write to the outputs.

Thank you very much for your quick answer. I think we can close this issue.

closed 2022-01-07 by davidDS96