OpenEtherCATsociety/SOEM #586: When can the entry port be other than zero?

sparksh, 2022-01-16:
In the file "ethercatdc.c" there is a function ecx_configdc with a comment "entryport is port with the lowest timestamp" - What configuration or situation could cause the entry port to be other than zero?

Thanks!

ArthurKetels, 2022-01-16:
An EtherCAT slave can have up to 4 ports. There is a strict order in routing. But depending on hot swap conditions some ports can be not connected, and sometimes ports can be connected in non-standard order. So for SOEM to figure out how the topology looks like, it needs to take into account the timestamp the test-packet has been given on the port. So physical port order does not necessarily needs to be the logical port order.

The simplest way to test this is to connect the master cable to the slave "OUT" port instead of the "IN" port. It will still work, but logical slave order is reversed. With multiple slaves a multitude of logical orders can be created.

sparksh, 2022-01-17:
Of course. This is necessary for supporting the redundant (reversed) ethernet cables. I forgot about that.
Thanks!

closed 2022-01-17 by sparksh