OpenEtherCATsociety/SOEM #588: Invalid configuration for syncmanager KEBA D3-DA

nick42h, 2022-01-24:
Hello SOEM-Community

I want to control two KEBA D3-DA 330/B-0314-Slaves each controlling three servo-motors. 

I started by connecting one slave and my plan is to send position-command-values to one of the servos. 
I managed to get the Slave to operational state (sending the Start-Up-List as Twincat does it, during the pre-operational-state).

But during the operational state I can not  reset an error code (the received status word can indicates an error bit and it can not be reset via the ControlWord). Further investigation (SDO: 0x2864) lead to the hint "Invalid configuration for syncmanager". 
With TwinCat the error does not appear and if I connect TwinCat after the Error got triggered via SOEM I can reset the error via the control word.

How can the configuration of the sync manager be invalid? What can I do to check the configuration?

Please find the ESI-file and WireShark-Logs attached. There is one WireShark-log from the SOEM-Startup and one from the TwinCat-Startup-Sequence.

ESI-File:
https://dataspace.keba.com/#/public/shares-downloads/u9EdIYcIeesGWxen
WireShark-Logs:
[wireshark_logs.zip](https://github.com/OpenEtherCATsociety/SOEM/files/7926345/wireshark_logs.zip)
klfyf0RYo7LBoOU3

ArthurKetels, 2022-01-24:
That is a complicated slave to configure. It is in the top 10 on my complicated list.
It uses 7 SM and 8 FMMU. 
There are at least 2 issues with the SM configuration, and 2 with the FMMU configuration.

SM:
1. SOEM reads the SM configuration from the SII (eeprom). There is a problem with the SM.activate register settings. You would expect this byte to be 0 or 1 (disabled or enabled) but it reads 3 (enabled+repeat request). This could trigger 0x2864. You could patch SOEM to clear the 2nd bit of SM.activate. It should never be set for configuration.
2. The SM length is first read from SII, but if CoE is available it reads the mapping there to calculate the length at run-time. It then over-writes the value found in SII. The problem is that during the CoE read of object 0x1c00 the response is for 7 SM but the size byte is only 5. So SOEM only tries to read the first 5 SM and over-writes the SII data. The last SM 6 and 7 mappings are never read and therefore not over-written. But they should as the actual length is zero and not 140. So both SM 6 and 7 have the wrong length. This could trigger 0x2864.

Both reasons could trigger the invalid configuration for syncmanager error. Both are small bugs in the slave SII/firmware. But you could work around it.

FMMU:
1. The algorithm that SOEM uses to attach FMMU's to SM is optimized for the least number of FMMU. However in the ESI file 8 FMMU's are defined for 6 SM. SM 4 is mapped to 3 FMMU's with the same values. As far I can see this has no meaning and just waists FMMU's. On the other hand this should not hurt the operation. 
2. In ethercatmain.h EC_MAXFMMU is defined as 4. You should modify this to 8 to be able to config.all FMMU's. At the moment not all FMMU's can be mapped.

For reference in the wireshark files:
CoE SM assignment response in SOEM, line 784
SM mappings in SOEM , lines 1185+
SM mappings in TwinCAT, lines 10249+

nick42h, 2022-01-26:
Thank you for the quick support. 
We finally fixed it by setting the length of the syncmanagers, that are relevant for the FSoE-datas, to zero.
Surprisingly we did not need to change EC_MAXFMMU.

closed 2022-01-26 by nick42h