OpenEtherCATsociety/SOEM #589: Dunkermotoren + Turck Ethercat Issue

kamilndbka, 2022-02-06:
Hello,
Guys - i need Your help with PDO implementation for Dunkermotoren BG95 dPro.
I am trying to connect 1 Turck BL20 I/O module and 2 Dunkermotoren BG95 dPro motors by PDO EC.
I am using ROS, but it shouldn't be a problem.

Currently:
I implemented initialization for all 3 devices, and all are going operation without any problem.
I already implemented PDO communication with Turck BL20 Ethercat I/O module (couple I/O, encoder, analog modules). The I/O module has internal configuration that can't be changed by TwinCat - the only issue was to read all IO area by Twincat and implement it into structure of code. 
I implemented SDO communication with dunkermotoren motors, but it is to slow to use it (approx. 6ms for each read/write register).
For Turck IO it's 1ms for read write all IO map.

I read tutorial.txt and Ethercat documentation but I have got couple issues:

1. Initialization (shown in tutorial) gives:

[ INFO] [1644170658.758096410]: Number of servos enabled: 2
[ INFO] [1644170658.758877125]: EtherCAT initialization...
[ INFO] [1644170658.781393731]: ec_init on enp0s25 succeeded
[ INFO] [1644170658.852157596]: 3 slaves found and configured
[ INFO] [1644170658.852257957]: Slave: 1 has CA?: Yes 

[ INFO] [1644170658.852336606]: Slave: 2 has CA?: No 

[ INFO] [1644170658.852376568]: Slave: 3 has CA?: No 

[ INFO] [1644170659.447571984]: Slave: 1
Name: BL20-E-GW-EC
Output size: 224 bits
Input size: 288 bits
State: 2
Delay: 0
Has DC: 
[ INFO] [1644170659.447681428]: Slave: 2
Name: Integrated drives
Output size: 0 bits
Input size: 0 bits
State: 2
Delay: 0
Has DC: 
[ INFO] [1644170659.447774142]: Slave: 3
Name: Integrated drives
Output size: 0 bits
Input size: 0 bits
State: 2
Delay: 0
Has DC: 

As You can see Slave 1 (turck i/o) provide output/input memory sizes, but servos (slave 2, 3) not. Slave 2 and 3 doesn't have CA.

The ESI file (and twincat) show that:
0x1a00 has 13 bytes to read;
0x1600 has 9 bytes to write
No idea why Input and Output size is 0.

2. 
I've tries configuration shown in #300 - only the part of setting 1c12 and 1c13, but is doen't changed anything. 

int BG95ProPDOSetup(uint16 slave)
{
    int retval=0;
    uint16 u16val;
    //RxPDO
    uint16 map_read = 0x1600;
    //TxPDO
    uint16 map_write = 0x1a00;
        
    retval +=  ec_SDOwrite(2, 0x1c12, 0x00, false, sizeof(map_read), &map_read, EC_TIMEOUTSAFE);
    retval += ec_SDOwrite(2, 0x1c13, 0x00, false, sizeof(map_write), &map_write, EC_TIMEOUTSAFE);

    return 1;
}

[Code.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8011038/Code.zip)
[ESI.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8011039/ESI.zip)

Can You help with this issue? I don't know what should I check more.

nakarlsson, 2022-02-08:
Post a Wireshark of the startup.
If not using CA solves your problem, why not just stick to that? CA or not CA have caused a lot of issues in the past for me.
Also, why not add a check of the SDO write operations to verify it was a success?

1. The mapping Objects are UINT32, <index:16><subindex:8><size:8>, always is.
2. The Syncmanager objects 1c12,1c13... hold active mappings objects. 0x1600-0x17FF ...
3. 1C12 : 00 is number of entries UINT8
4. 1C12 : >0 is the actual mappings objects.
5. SOEM iterate the syncmanager objects to find the mapping.
6. The size of the mapping objects ...<size:8> are added together and becomes the total size that the SM and FMMU are programmed with.
7. The P2SHook allows you, if the slave admits it, to program the syncmanager and mapping objects before SOEM does the final calculation to be programmed.

All info above can be found in the specifications for EtherCAT ETG1000 and ETG1020 for details on CA.

kamilndbka, 2022-02-08:
There is no possibility to read / write 0x16XY and 0x1AXY. All time Abort connection and SDOwrite return = 0;

`
int GatewayNode::ethercatInit()
{
    ROS_INFO_STREAM_COND(debug_mode,"EtherCAT initialization...");

    /* initialise SOEM, bind socket to ifname */
    if (ec_init(interface_name.c_str()))
    {
	    ROS_INFO_STREAM_COND(debug_mode,"ec_init on " << interface_name << " succeeded");
        /* find and auto-config slaves */

        /** network discovery */
        if ( ec_config_init(false) > 0 ) //sprawdza czy siec wstała, czy udało sie skonfiguraowac port, sprawdzic!
        {
	        ROS_INFO_STREAM_COND(debug_mode,ec_slavecount << " slaves found and configured");
			
            for (int i=1; i<=ec_slavecount; i++) 
            {
                if(ec_slave[i].CoEdetails & ECT_COEDET_SDOCA)
                {
                    ROS_INFO_STREAM_COND(debug_mode,"Slave: " << i << " has CA?: Yes \n");
                }else
                {
                    ROS_INFO_STREAM_COND(debug_mode,"Slave: " << i << " has CA?: No \n");
                }				
                /** CompleteAccess disabled for Elmo driver */
                ec_slave[i].CoEdetails ^= ECT_COEDET_SDOCA;
            }
            ec_statecheck(0, EC_STATE_PRE_OP,  EC_TIMEOUTSTATE); //sprawdza stan slavów
            
            //PDO Config
            int ret = 0;

            uint8_t value8 = 0;
            uint32_t value32 = 0;
            int l = sizeof(value32);

            //set 0x1600 in 0x1c12

      
            //RxPDO
            value8 = 0;
            ret +=  ec_SDOwrite(2, 0x1c12, 0x00, false, sizeof(value8), &value8, EC_TIMEOUTSAFE);
            ROS_INFO_STREAM_COND(debug_mode," 1c12:00, 0 ret: "<<ret);
            uint16_t map_read = 0x1607;
            ret +=  ec_SDOwrite(2, 0x1c12, 0x01, false, sizeof(map_read), &map_read, EC_TIMEOUTSAFE);
            ROS_INFO_STREAM_COND(debug_mode," 1c12:01, 1606 ret: "<<ret);
            value8 = 1;
            ret +=  ec_SDOwrite(2, 0x1c12, 0x00, false, sizeof(value8), &value8, EC_TIMEOUTSAFE);
            ROS_INFO_STREAM_COND(debug_mode," 1606:00, 1 ret: "<<ret);

            //TxPDO
            value8 = 0;
            ret +=  ec_SDOwrite(2, 0x1c13, 0x00, false, sizeof(value8), &value8, EC_TIMEOUTSAFE);
            uint16_t map_write = 0x1a07;
            ret += ec_SDOwrite(2, 0x1c13, 0x01, false, sizeof(map_write), &map_write, EC_TIMEOUTSAFE);
            value8 = 1;
            ret +=  ec_SDOwrite(2, 0x1c13, 0x00, false, sizeof(value8), &value8, EC_TIMEOUTSAFE);
            ROS_INFO_STREAM_COND(debug_mode," all: "<<ret);     

            value8 = 0;
            ret += ec_SDOwrite(2, 0x1607, 0x00, false, sizeof(value8), &value8, EC_TIMEOUTRXM);
            ROS_INFO_STREAM_COND(debug_mode," 1606:00, 0 ret: "<<ret);
            value32 = 0x6040;//0010;
            ret += ec_SDOwrite(2, 0x1607, 0x01, false, sizeof(value32), &value32, EC_TIMEOUTRXM);
            ROS_INFO_STREAM_COND(debug_mode," 1606:01, 6040 00 10 ret: "<<ret);
            value8 = 1;
            ret += ec_SDOwrite(2, 0x1607, 0x00, false, sizeof(value8), &value8, EC_TIMEOUTRXM);
[... 46 lines cut ...]
            ec_writestate(0);
            int chk = 200;
            /* wait for all slaves to reach OP state */
            do
            {
                ec_send_processdata();
                ec_receive_processdata(EC_TIMEOUTRET);
                ec_statecheck(0, EC_STATE_OPERATIONAL, 50000);
                ec_statecheck(1, EC_STATE_OPERATIONAL, 50000);
                ec_statecheck(2, EC_STATE_OPERATIONAL, 50000);
            }
            while (chk-- && (ec_slave[0].state != EC_STATE_OPERATIONAL) && (ec_slave[1].state != EC_STATE_OPERATIONAL) && (ec_slave[2].state != EC_STATE_OPERATIONAL) );

            
            return 1;

        }

    }
    return 0;
}

`
Please find attached wireshark logs. Look at 1607 and 1a07 data...
[ECAT_Initialization.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8026536/ECAT_Initialization.zip)

ArthurKetels, 2022-02-08:
I guess you need to ask your servo drive vendor for a firmware update. It advertises CA (Complete Access) in the EEPROM but in the ESI file it is not available. Clearly from the try SOEM makes it is only partially available, give the error message after the read of SDO 0x1c00: `{0x06010004, "SDO Complete access not supported for variable length objects" }`. When SOEM then retries with legacy access the error is : `{0x06020000, "The object does not exist in the object directory" }`. This is even worse because this is a mandatory object, and the slave conformance test will fail for this slave. SOEM can then not continue iterating over the slave mapping so it does not set any TxPDO or RxPDO in the IOmap, nor programs the SM and FMMU of the slave.

You could try to work around the issues of this slave firmware, SOEM is very flexible, but that is a task more suited to experienced EtherCAT programmers.

kamilndbka, 2022-02-08:
The funny thing about it is that in knowledge base (manual) of servo drive, vendor gives info shown below for each parameter (PDO mapping: YES).
Also the servo is configurable from Twincat - the communication works and data are valid.
I don't know what to do with it 

[image]

ArthurKetels, 2022-02-09:
We are talking about different issues. I take a top-down approach. Object 0x1c00 is where the mapping of TxPDO and RxPDO starts. When that is not working properly it is no use to continue. Your remark is about the PDO assignment and PDO mapping. They come later. But indeed something is fishy there too. The proper way to investigate this is to make Wireshark traces of both TwinCAT and SOEM start-up sequences for the same intended drive configuration (compare apples with apples). Then see where the differences are. Remodel SOEM to mimic TwinCAT.

Often beginning EtherCAT slave vendors tend to check their firmware against TwinCAT and tweak until it works. This leaves other master implementations like SOEM untested. So bugs remain. There is also an official (and mandatory) test tool (Conformance Test Tool) that would pick out these issues instantly. How it is possible that you have a device that will fail that mandatory test is a mistry to me.

nakarlsson, 2022-03-08:
@kamilndbka , can we close this issue?

nakarlsson, 2022-03-30:
Close due to inactivity

closed 2022-03-30 by nakarlsson

kamilndbka, 2022-04-27:
Dear,
I tried to work once again on the SOEM and Dunkermotoren servo.
I implemented once again the init code with ROS.

Please find code shown below:
```

int GatewayNode::ethercatInit()
{
   
   /* ROS_INFO_STREAM_COND(debug_mode,"EtherCAT initialization...");

    /* initialise SOEM, bind socket to ifname */
    if (ec_init(interface_name.c_str()))
    {
	    ROS_INFO_STREAM_COND(debug_mode,"ec_init on " << interface_name << " succeeded");
        /* find and auto-config slaves */

        /** network discovery */
        if ( ec_config_init(false) > 0 ) //sprawdza czy siec wstała, czy udało sie skonfiguraowac port, sprawdzic!
        {
	        ROS_INFO_STREAM_COND(debug_mode,ec_slavecount << " slaves found and configured");usleep(100);

        }  

        for (int i=0; i<ec_slavecount; i++) 
            {
                ROS_INFO_STREAM_COND(debug_mode,"STATE: " << ec_slave[i].state);usleep(100);
            }

	
            ec_statecheck(0, EC_STATE_PRE_OP,  EC_TIMEOUTSTATE); //sprawdza stan slavów
            ec_statecheck(1, EC_STATE_PRE_OP,  EC_TIMEOUTSTATE); //sprawdza stan slavów
            ec_statecheck(2, EC_STATE_PRE_OP,  EC_TIMEOUTSTATE); //sprawdza stan slavów

            for (int i=0; i<ec_slavecount; i++) 
            {
                ROS_INFO_STREAM_COND(debug_mode,"a STATE: " << ec_slave[i].state);usleep(100);
            }
            
            ec_slave[1].PO2SOconfig=BG95ProPDOSetup;usleep(100);
            //ec_slave[3].PO2SOconfig=BG95ProPDOSetup;usleep(1000);
            //BG95ProPDOSetup(2);usleep(100);

            /** if CA disable => automapping works */
            ec_config_map(&turck.IOmap);usleep(1000);

            for (int i=0; i<ec_slavecount; i++) 
            {
                ROS_INFO_STREAM_COND(debug_mode,"b STATE: " << ec_slave[i].state);
            }
                     

            for (int i=1; i<=ec_slavecount; i++) 
            {
                ROS_INFO_STREAM_COND(debug_mode,"Slave: " << i << "\nName: " << ec_slave[i].name 
                << "\nOutput size: " << ec_slave[i].Obits << " bits\nInput size: " 
                << ec_slave[i].Ibits << " bits\nState: " << ec_slave[i].state 
                << "\nDelay: " << ec_slave[i].pdelay << "\nHas DC: " << ec_slave[i].hasdc);
            }
            //teraz urzadzenia gotowe do pracy, są w trybie initial
            ROS_INFO_STREAM_COND(debug_mode,"Slaves mapped, state to SAFE_OP.");
            /* wait for all slaves to reach SAFE_OP state */
            ROS_INFO_STREAM(" ecstate0: "<<(int)ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE));
            ROS_INFO_STREAM(" ecstate1: "<<(int)ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE));
            ROS_INFO_STREAM(" ecstate2: "<<(int)ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE));
            //ec_statecheck(1, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);
            //ec_statecheck(2, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);
            for (int i=0; i<ec_slavecount; i++) 
            {
[... 310 lines cut ...]
            ret=0;
            value8 = 0x02;
            ret +=  ec_SDOwrite(slave, 0x1c13, 0x00, false, sizeof(value8), &value8, 10*EC_TIMEOUTSAFE);usleep(300);
            ROS_INFO_STREAM(" 1c13:00, 0 ret: "<<(int)ret);

           

            ec_SDOread(slave,0x1c13, 0x00, 0, &kk, &value8,10*EC_TIMEOUTSAFE );
            ROS_INFO_STREAM(" 1c13:00 read , 1 ret: "<<(int)value8);
            ec_SDOread(slave,0x1c12, 0x00, 0, &kk, &value8,10*EC_TIMEOUTSAFE );
            ROS_INFO_STREAM(" 1c12:00 read , 1 ret: "<<(int)value8);

    return 1;
}
```

All BG95ProPDOSetup writes are corretly provided - all rets are 1. 
The problem (in my opinion) begins with ec_config_map cause there is no possibility to send data to 1c00.
Then servo can't get into OP state - last state is 6 or 22 and last do_while loop is done 200 times.
I checked 1C32:1 - mode is free run (return 0) and 1C32:5 - ret is 5 (free run and dc_mode). same for 1c33.
I also checked is PDO datas are correcttly assigned and they are (send PDO's=received)

Do You have any ideas what should be done? is there any other way to do it ?

[a.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8575560/a.zip)