OpenEtherCATsociety/SOEM #590: LAN9254 with SOEM

MikeCalugaru, 2022-02-11:
Hello,

I'm wondering if anyone has managed to configure the SOEM library to work with Microchip's LAN9254 EtherCAT Slave Controller. I'm looking for any basic working example, such as a blinking LED that works on the LAN9254 Evaluation board (EV57N07A). Any advice or pointers on how to change the examples in the SOEM to work with LAN9254 are much appreciated.

Thank you

nakarlsson, 2022-02-11:
Hi,
given your LAN9254 is working properly, the examples should work out of the box.
At least to make sure it is alive and have basic functionallity in place.

MikeCalugaru, 2022-02-11:
Hi @nakarlsson,

I am able to discover the LAN9254 as a slave, initialize it, do round trips, and set it on OP mode; all this works fine. However writing to the slave's output data register or reading its input data register does not seem to work. Setting the bits in the output register should lite up the LED on the evaluation board (EV57N07A) which does not happen (my board is configured 16 IN and 16 OUT).

I've tried to use the functions below from the examples to set the bits in the output register or write a whole digit with no success. I also tried to run an adapted version of the code (for the LAN9254) in the test->rtk folder; also did not work. 

Interestingly enough, using pysoem, which is a python wrapper around SOEM (https://github.com/bnjmnp/pysoem) works just fine; I am able to write and read with no issues. 
 
`void set_output_bit (uint16 slave_no, uint8 module_index, uint8 value)
{
   /* Get the the startbit position in slaves IO byte */
   uint8 startbit = ec_slave[slave_no].Ostartbit;
   /* Set or Clear bit */
   if (value == 0)
      *ec_slave[slave_no].outputs &= ~(1 << (module_index - 1 + startbit));
   else
      *ec_slave[slave_no].outputs |= (1 << (module_index - 1 + startbit));
}   `

`void set_output_int16 (uint16 slave_no, uint8 module_index, int16 value)
{
   uint8 *data_ptr;
   /* Get the IO map pointer from the ec_slave struct */
   data_ptr = ec_slave[slave_no].outputs;
   /* Move pointer to correct module index */
   data_ptr += module_index * 2;
   /* Read value byte by byte since all targets can handle misaligned
    * addresses
    */
   *data_ptr++ = (value >> 0) & 0xFF;
   *data_ptr++ = (value >> 8) & 0xFF;
}`

nakarlsson, 2022-02-12:
Did you test simple_test? If not, use that and simply add 

….

/* set all outputs to ’1’ for first slav on EtherCAT link */
/* Assume first slave got 16bit output */
uint16_t value = 0xffff;
memcpy( ec_slave[1].outputs, &value, sizeof(value));

ec_send_processdata();

nakarlsson, 2022-03-08:
@MikeCalugaru , can we close this issue?

MikeCalugaru, 2022-03-10:
Hi @nakarlsson,

If you want to close the issue, I'm happy for you to close it. I'm afraid I was unable to establish proper communication with the LAN9254 chip using the SOEM library (unfortunately could not spend much time to fiddle with the code). I was able to make the pySOEM library work "out of the box". In the end I reverted to using Beckhoff TwinCAT tool-stack to do my testing. TwinCAT/Acontis rely on ESI/ENI files to do all the configurations which is the preferred approach (where I work).

If you could add a working example for the LAN9253/4/5 family of ESC chips (using Microchip's evaluation boards) that would be fantastic. The SOEM only has examples using the Beckhoff ET1100 ASIC chip. Thank you.

nakarlsson, 2022-03-12:
Ok. 

I’m sure a proper LAN9253 implementation will work out of the box with Vanilla SOEM aswell. I’ve used the LAN9252 several times.

closed 2022-03-12 by nakarlsson