OpenEtherCATsociety/SOEM #593: Running SOEM in multi-thread environment

gvcode1, 2022-02-18:
Hi Team,

On multi-thread environment like Linux, we saw the example code use main task to do the initialization and then enter the cyclic data exchange.

Normally it should works fine, but if we want to read/write SDO in background task or low priority task, it may meet the problem that higher cyclic data exchange task interrupt SDO read/write process. Normally SDO read/write will send additional frame and wait the feedback frame until timeout.

So I guess the mutex should be add in order to protect the SDO read/write will not be interrupt. If we bring the mutex to protect the SDO read/write process, then it may affect the realtime performance of cyclic data exchange task.

Am I correct? 

Or we only need to protect the Ethernet Frame Send/Receive API?

If we only protect the Ethernet Frame Send/Receive API call, does SOEM can correctly handle the EtherCAT frame sequence, for example: When SDO command has been sent, but not received yet, then cyclic data task interrupt the SDO, and send the PDO frame, then cyclic data task wait for the PDO frame back, but now the SDO frame has come back first, then PDO frame comming, does SOEM will check it automaticly?

Thanks & Best Regards,

nakarlsson, 2022-02-19:
It is handled by SOEM, you can run SDO read/write in a background task. The nicdrv.c  handled the low level locking. Via a frame index SOEM known what frame goes where.

gvcode1, 2022-02-22:
Understoold, thanks a lot for your support!

Btw:Due to the memory limitation, we limited max buffer to 8
#define EC_MAXBUF          8//16

As I saw that SOEM always will send and receive the package in blocking mode, and we only handle 1 SDO read and write in each cycle call, do you think 8 buffer size is enough for the application?

Thanks!

nakarlsson, 2022-02-22:
Yes, we have small systems using 4.

gvcode1, 2022-02-23:
Thanks a lot for your support! Understood!

closed 2022-02-23 by gvcode1