OpenEtherCATsociety/SOEM #599: Problems with MSI-FB-Ethercat

Haegar88, 2022-03-10:
Hello,
I need some help to integrate a Leuze MSI-FB-Ethercat Gateway on a BigEndian machine. I use ecx_config_init and ecx_config_map_group.
In the startup function it is not possible to access objects like 1C00, 1C12, 1C13 (SDO slave:1 index:1c00.00 error:06020000 The object does not exist in the object directory).
I then tried to access the module with TwinCat3. However, this only worked after I added the corresponding ESI/XML file from Leuze. The attempt to find out the differences in the configuration with Wireshark failed, as they are completely different.
Then I became aware of the SII interface and found out that some of the data read from the EEprom was mixed up (highbyte, lowbyte...) Here I removed endian macros and now get plausible values for the individual categories (String, General, FMMU and SM). Unfortunately, the category PDO does not return a hit and therefore no IBytes and OBytes are entered. I once dumped the EEprom and indeed this category is missing there. I then looked at the EEprom with TwinCat3 and there is much more there. Apparently the data is transferred from the XML to the EEprom. The PDO category is also there after the SM category. What else can I do? Does somebody know advice? The configuration fails (IOMap failed) and the device remains in status 17. I have also attached the values of the slave structure after cofiguring and the two wireshark records.
Greetings

[wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8223527/wireshark.zip)
[image]
[image]
[SlaveStructure.TXT](https://github.com/OpenEtherCATsociety/SOEM/files/8223368/SlaveStructure.TXT)

ArthurKetels, 2022-03-11:
Yeah, another (not so) great slave implementation. I do not know if it is possible to save the TwinCAT SII information to EEPROM? Then the slave should configure automatically with SOEM. Another option is to have the PO2SOconfigx hook enabled for this slave and then manually set the FMMU parameters, and the Ibits and Obits. There is a little trick, if you also set configindex of this slave to something like 9999 then the COE mapping will be skipped and the current SM and FMMU settings will be used to configure the slave.

nakarlsson, 2022-03-30:
@Haegar88 , can we close this issue?

nakarlsson, 2022-11-17:
Close due to inactivity

closed 2022-11-17 by nakarlsson