OpenEtherCATsociety/SOEM #600: Rexroth XB-EC-12 COUPLER WKC Problem

DamonTTSA, 2022-03-15:
Good day

I am using SOEM to communicate with a Rexroth XB-EC-12 COUPLER, which is connected to a Rexroth digital input module and a Rexroth digital output module (ctrlX IO).
I calculate the expected WKC like the SOEM tutorial shows : 
[image]

However, this calculation does not work for this hardware setup.
The received WKC is 3, but the expected WKC is 5.
By removing the modules one-by-one, I could see that the DO module increments the WKC by two, the DI module increments the WKC by one, and the coupler does not increment the WKC. Clearly SOEM expects that the coupler increments the WKC by two.
I have enquired at Rexroth, but their contact person just tested the modules with their own ctrlX master and does not know much about EtherCAT.

I have used this calculation before, with a Beckhoff coupler, DI module and DO module. In this case, the expected and received WKCs matched, so SOEM worked as expected. I don't have the modules anymore, so I can't check what the WKC was, but if I remember correctly, it was 5 (so coupler incremented by 2).

My question is: what is the protocol standard for an EtherCAT bus coupler? Does it usually have an 'output', causing it to increment the WKC by two?
I am wondering if the bus coupler I have is faulty, or whether it doesn't conform to the EtherCAT standard properly.

ArthurKetels, 2022-03-15:
The working counter  increment is a function of an activated FMMU by the LRW packet. Each read FMMU increments the workcounter by one and each write FMMU increments the workcounter by two. Slaves have up to 8 FMMU's available for PDO. Depending on the FMMU configuration the workcounter could be incremented up to 16.

To have a more meaningful answer please post  a wireshark capture of the configuration sequence. The ESI file would also be welcome. Mostly SOEM gets it right but there have been situations in the past that resulted in wrong WKC calculations. This could happen when a FMMU is configured but not activated, or when the FMMU is configured and activated but zero length.

ljvmaanen, 2022-03-18:
[Bosch_IO.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8304972/Bosch_IO.zip)

Hello Arthur,

Thanks for your reply. I did some tests on the Rexroth hardware connected to SOEM on a PC and added the Wireshark capture file and "slave_info" output. The actual WKC is 3 and the expected WC is 5. This is the same result as with our embedded controller.

Damon is on the plane back to South Africa. He will send you the ESI file later. 

Kind regards,

John

ArthurKetels, 2022-03-18:
Thanks for the file. It makes the problem more clear. Tl&Dr : The eeprom of the XI211116 has the wrong SM type for the outputs, it defines 3-buffer and should be 1-buffer. Modify the eeprom and you solved the problem.

Longer explanation:
The output module defines the SM and FMMU like below.
```
 SM0 A:0f00 L:   1 F:00090044 Type:3
 SM1 A:0f01 L:   1 F:00090044 Type:3
 FMMU0 Ls:0000000a Ll:   2 Lsb:0 Leb:7 Ps:0f00 Psb:0 Ty:02 Act:01
```
The slave is of a CPU less direct output type. In the ESC the address range for digital outputs is 0x0f00 and further. A write to this register will set the output pins on the IC directly. This slave has 16 output pins and therefore uses address 0x0f00 and 0x0f01. 

A simple solution would be to have no SM and only a FMMU that is mapped to both registers. And this works fine. However there is one problem that when the slave is in OP and the master communication fails there is no watchdog to clear the outputs after a timeout. There is no  CPU to monitor the data and control the protocol. 

A SM can be programmed to have a watchdog and clear the register. But it is not possible to use the regular 3-buffer SM because that will occupy 3 address ranges to be used as swap buffers. The direct output register 0x0f00 is however no normal RAM and can not be used for 3-buffer. There is a special exception for SM, when the size is 1 the SM is disabled for buffering but the watchdog is still active. That is why we see two SM defined of 1 byte each, 0x0f00 and 0x0f01. 

Now comes the reason why the WKC does not update.  A 3-buffer only transfers data from the master to slave when the SM is active and there is an empty buffer to write to. This is not the case here. The result is that the FMMU write fails and the WKC from the LRW packet is not increased. 

If we use a 1-buffer for this SM then the write of the FMMU will still succeed because the target register and the buffer are on the same address. It is a bit of a hack but it will work.

ljvmaanen, 2022-03-19:
Hello Arthur,

Thanks for your fast response again. If I understand correctly, the problem is caused by the Rexroth module and not SOEM. We will contact them about this, as it is kind of a prototype for them as well. If they are not going to change the EEPROM information, we will try your workaround. 

Thanks again. Your perfect support is of great value to us.

Kind regards,

John

ArthurKetels, 2022-03-19:
Turns out I was wrong with my statement above. I just tested a Beckhoff digital output module. It has the same SM configuration as yours. It works fine with those settings. So the 1 byte SM will disable the 3-buffer mechanism too, just leaves the watchdog active.

Re-reading the top post it is clear that it is not the DO slave causing trouble but the coupler slave wit CPU. Then it looks like a case of disabled SM by the slave firmware. A read (FPRD) of the SM registers while in OP will show if this is the case or not.

DamonTTSA, 2022-03-21:
Hi Arthur

Thanks again for your quick responses!

Please find the ESI file for the Rexroth EC coupler attached.

[BoschRexroth_ctrlX_IO_XB-EC-12_ESI.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8313544/BoschRexroth_ctrlX_IO_XB-EC-12_ESI.zip)

ljvmaanen, 2022-03-21:
Hello Arthur,

Maybe I'm wrong, but I think it's strange that these parameters are configured as outputs (according to slave_info):

            PDO mapping according to CoE :
              SM2 inputs
                 addr b   index: sub bitl data_type    name
              SM3 outputs
                 addr b   index: sub bitl data_type    name
              [0x0000.0] 0x6000:0x01 0x10 UNSIGNED16   UP Voltage
              [0x0002.0] 0x6000:0x02 0x10 UNSIGNED16   UP Current
              [0x0004.0] 0x6010:0x01 0x10 UNSIGNED16   UL Voltage
              [0x0006.0] 0x6010:0x02 0x10 UNSIGNED16   UL Current
              [0x0008.0] 0x6020:0x01 0x01 BOOLEAN      UP Undervoltage
              [0x0008.1] 0x6020:0x02 0x01 BOOLEAN      UP Overvoltage
              [0x0008.2] 0x6020:0x03 0x01 BOOLEAN      UP Overcurrent
              [0x0008.3] 0x6020:0x04 0x01 BOOLEAN      UL Undervoltage
              [0x0008.4] 0x6020:0x05 0x01 BOOLEAN      UL Overvoltage
              [0x0008.5] 0x6020:0x06 0x01 BOOLEAN      UL Overcurrent
              [0x0008.6] 0x6020:0x07 0x0A Type 0x0000  SubIndex 007
  
I would expect them to be inputs (data to the master). 
  
Kind regards,

John

ArthurKetels, 2022-03-21:
Thanks for the data. Now it is clear. Indeed the problem is with the coupler station. Object 0x1c00 has the byte order reversed in the CA. This is a known bug in the Beckhoff SSC. As you pointed out the inputs and outputs are reversed.
The output from slaveinfo:
```
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   0 F:00000064 Type:4
 SM3 A:1400 L:  10 F:00010020 Type:3
```
As you can see SM2 is type 4 (inputs) and SM3 is type 3 (outputs). This should be reversed.

```
 Index: 1c00 Datatype: 0005 Objectcode: 08 Name: Sync manager type
  Sub: 00 Datatype: 0005 Bitlength: 0008 Obj.access: 0007 Name: SubIndex 000
          Value :0x04 4
  Sub: 01 Datatype: 0005 Bitlength: 0008 Obj.access: 0007 Name: SubIndex 001
          Value :0x02 2
  Sub: 02 Datatype: 0005 Bitlength: 0008 Obj.access: 0007 Name: SubIndex 002
          Value :0x01 1
  Sub: 03 Datatype: 0005 Bitlength: 0008 Obj.access: 0007 Name: SubIndex 003
          Value :0x04 4
  Sub: 04 Datatype: 0005 Bitlength: 0008 Obj.access: 0007 Name: SubIndex 004
          Value :0x03 3
```
Object 0x1c00 has the same problem. SM0 = 2, SM1 = 1, SM3 = 4 and SM4 = 3. This should be 1, 2, 3, 4. SOEM has some auto correct logic to reverse SM0 and SM1 to the correct values. But it can not do that for PDO SM. What the slave commands SOEM should strictly follow. There are some very complex slaves out there with up to 8 SM defined. There is no way SOEM could figure out the correct order by itself.

The solution is a bug fix in the slave firmware, see #401

ljvmaanen, 2022-03-21:
Thank you, Arthur. 

We will contact Rexroth about it. We are trying to test our Ethercat controller with "stable" slaves, but most of the time we find problems in the supplied hardware.

Kind regards,

John

closed 2022-03-22 by ArthurKetels