OpenEtherCATsociety/SOEM #604: Integrating CoE SDOs into cyclic PDO frames

DamonTTSA, 2022-04-04:
Good day

In our system, we have PDO cycles every 1ms. These PDO cycles are locked onto a DC reference clock.
We would like to access some CoE objects once these PDO cycles have started (in the OP state).
To do this, we need to append a CoE SDO datagram to the datagram that we are already sending, like if datagram ex.1 and datagram ex.2 below were in the same frame:
[image]

However, it looks like SOEM does not support this in its base form.
The call paths for sending SDO's and PDO's show that separate datagrams have to be used:

**Send PDO’s (IO data):**
void pdos_tx_rx(unsigned char DC_32_bits)
int ec_send_processdata(void)
int ec_send_processdata_group(uint8 group)
int ecx_send_processdata_group(ecx_contextt *context, uint8 group)
static int ecx_main_send_processdata(ecx_contextt *context, uint8 group, boolean use_overlap_io)
int ecx_outframe(ecx_portt *port, uint8 idx)

**Send CoE SDO’s**
int ec_SDOwrite(uint16 Slave, uint16 Index, uint8 SubIndex, boolean CA, int psize, void *p, int Timeout)
int ecx_SDOwrite(ecx_contextt *context, uint16 Slave, uint16 Index, uint8 SubIndex,      boolean CA, int psize, void *p, int Timeout)
int ecx_mbxsend(ecx_contextt *context, uint16 slave,ec_mbxbuft *mbx, int timeout)
int ecx_FPWR(ecx_portt *port, uint16 ADP, uint16 ADO, uint16 length, void *data, int timeout)
int ecx_srconfirm(ecx_portt *port, uint8 idx, int timeout)
int ecx_outframe(ecx_portt *port, uint8 idx)

Is there a way to integrate the CoE mailbox communication into the PDO cycle frame so that we do not disrupt our DC?

Any advice would be much appreciated.

Kind regards
DamonTTSA

nakarlsson, 2022-04-07:
We have it in the roadmap. There are 2 approaches.
1. To add  multi datagram support
or
2. Make the SDO write/read non-blocking eg. easier to run in sequence with the PDO handling in the process data.
Or both, they are not mutually exclusive.

We've run test versions of 2. but it is still in the feature to get into master.

DamonTTSA, 2022-04-07:
Hi @nakarlsson 

Thank you for your response!

That's great news.
When can we expect it to be merged into the master?

Kind regards
DamonTTSA

nakarlsson, 2022-04-07:
I got no date, it is work in progress. You know we do most work here for free or at our companies "spare time" or based on paying customer projects.

closed 2022-04-07 by DamonTTSA