OpenEtherCATsociety/SOEM #606: The issue of 'incorrect wkc' !?

tom9672, 2022-04-07:
I got the error of incorrect wkc (checked the actual wkc is 3, and the expected wkc is 9, what is the mean 3 and 9?). 
The problem not occur very often, sometimes it happens while I click a button to do one more action while the PDO loop, create a thread to give more cmd to device, or read SDO, or PDO. The incorrect wkc happens like around  50 times I call some functions.

I use a Raspberry with its onboard Ethernet port, and use a 5.15 rt-kernel. 

What might lead to the 'incorrect wkc'?

Thanks in advance.

nakarlsson, 2022-04-07:
9->3 means the slaves have entered SAFEOP + Error. Only inputs get updated.

Most likely you delay tihe PDO thread long enough for the slaves Watchdogs to tripp. Eg. no frames are sent for 100ms (default Watchdog timeout value) or what value you set.

tom9672, 2022-04-15:
Solve the problem now, one thread takes almost 100% cpu, makes the delay of pdo process thread. Thank you

closed 2022-04-15 by tom9672