OpenEtherCATsociety/SOEM #607: Can't get EPROM data from epromtool

michaelschem, 2022-04-13:
I'm trying to get epromtool to get EPROM from my ethercat slave.

I've tried the following:

```
./eepromtool en8 1 -ri 0
SOEM (Simple Open EtherCAT Master)
EEPROM tool
ec_init on en8 succeeded.
1 slaves found.
Slave 1 data
 PDI Control      : 0E08
 PDI Config       : EE02
 Config Alias     : 0000
 Checksum         : 007A
   calculated     : 007A
 Vendor ID        : 0000060A
 Product Code     : ED310001
 Revision Number  : 00000001
 Serial Number    : 00000000
 Mailbox Protocol : 000C
 Size             : 000F = 2048 bytes
 Version          : 0001

Total EEPROM read time :654ms
End, close socket
End program
```

I'm assuming I'm using the tool wrong, but how do I get something at the first memory location?

ArthurKetels, 2022-04-13:
You now should have a file named "0" with the eeprom content in Intel Hex format.

michaelschem, 2022-04-13:
Indeed. Maybe this should be added to the usage?

closed 2022-04-13 by michaelschem

ArthurKetels, 2022-04-13:
Well.... It is mentioned in the usage, and is also a very common pattern for command line tools.
`      printf("Usage: eepromtool ifname slave OPTION fname|alias\n");`
The fname bit should be clear enough. Where else should the data come from when writing the eeprom?

michaelschem, 2022-04-14:
oh 🤦🏼 , didn't realize `fname` was for file name.  What is alias?

ArthurKetels, 2022-04-14:
Mandatory for each EtherCAT slave is a mechanism to permanently store a slave address. This is called the slave alias address (16 bit). Most commonly this is stored in the eeprom at a fixed memory address. The eepromtool allows to write this alias address in the eeprom without overwriting any other locations.

michaelschem, 2022-04-14:
I see, so fname is for ready alias is for write.