OpenEtherCATsociety/SOEM #608: Is there a way to use an EtherCATInfo  XML with SOEM

michaelschem, 2022-04-14:
I'm trying to interface with a EtherCat slave and the vendor has given me an EtherCATInfo XML (attached). Can I use this with SOEM to map values to named entities on the device
[ESTUN-ProNet plus-V208B0.xml.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8485812/ESTUN-ProNet.plus-V208B0.xml.zip)
?

ArthurKetels, 2022-04-14:
Yes off course. For example:
```
							<Object>
								<Index>#x6076</Index>
								<Name>Motor_Rated_Torque</Name>
								<Type>UDINT</Type>
								<BitSize>32</BitSize>
								<Flags>
									<Access>ro</Access>
									<Category>o</Category>
									<PdoMapping>RT</PdoMapping>
								</Flags>
							</Object>
```
This object (Motor rated torque) is read only uint32_t index 0x6076 subindex 0x00 and you can do :
`wkc = ec_SDOread(Slave, 0x6076, 0x00, FALSE, &rdl, &rdat, EC_TIMEOUTRXM);`

nakarlsson, 2022-10-18:
@michaelschem , can we close this issue?

michaelschem, 2022-10-18:
@nakarlsson I never got a chance to test this solution, but we can close it for now.

closed 2022-10-18 by michaelschem