OpenEtherCATsociety/SOEM #610: WKC wrong

jacckk, 2022-04-21:
Hi all, 

I'm trying to run simple_ng connected to a HBM TIM-EC EtherCAT module. The slave appears to go operational according to its status indicators but I get the following log from simple_ng.

`Initializing SOEM on '\Device\NPF_{B71CAD00-C2C1-434C-854F-2BEDF1A28A67}'... done
Finding autoconfig slaves... 1 slaves found
Sequential mapping of I/O... mapped 2O+42I bytes from 1 segments
Configuring distributed clock... done
Waiting for all slaves in safe operational... done
Send a roundtrip to make outputs in slaves happy... done
Setting operational state... all slaves are now operational
Iteration    1:     0 usec  WKC 0 wrong (expected 3)
All slaves resumed OPERATIONAL
Iteration    2:     0 usec  WKC 0 wrong (expected 3)
All slaves resumed OPERATIONAL
Iteration    3:     0 usec  WKC 0 wrong (expected 3)
All slaves resumed OPERATIONAL
Iteration    4:     0 usec  WKC 0 wrong (expected 3)
All slaves resumed OPERATIONAL
Iteration    5:     0 usec  WKC 0 wrong (expected 3)
All slaves resumed OPERATIONAL
Iteration    6:     0 usec  WKC 0 wrong (expected 3)
All slaves resumed OPERATIONAL
Iteration    7:     0 usec  WKC 0 wrong (expected 3)`

Here's a wireshark trace running simple_ng.
[simple_ng.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8529114/simple_ng.zip)

Any help on what to look for to debug this further would be appreciated.

ArthurKetels, 2022-04-21:
Can you run slaveinfo -map and paste the output here? 
There seems to be some timing issue with CoE in your wireshark log. SOEM does not wait for an answer from the slave when it requests object 0x1c00. It then continues to get the configuration data from eeprom, and this might be wrong.

jacckk, 2022-04-21:
Here's the output
```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{B71CAD00-C2C1-434C-854F-2BEDF1A28A67} succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:TIM EC
 Output size: 16bits
 Input size: 336bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000011d ID: 00000403 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:   2 F:00010064 Type:3
 SM3 A:1c00 L:  42 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   2 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000002 Ll:  42 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 07 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
End slaveinfo, close socket
End program

```
and heres the wireshark log
[slaveinfo.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8530814/slaveinfo.zip)

ArthurKetels, 2022-04-21:
Hmm, the Slave mailbox seems to be not active. That is why we do not get any CoE transfers. Can you post the ESI file of the TIM EC (I have to register to download it from the HBM website)? Possibly the mailbox SM config is not OK.

ArthurKetels, 2022-04-21:
Configuration of mailbox and PDO seem to be OK. Slave does not have any error state. It is just not possible to read or write the mailbox. All FPWR and FPRD to the mailbox return a workingcounter of zero. This means that the memory access in the slave was not accepted. As far as I can see nothing wrong with SOEM. Is there any other way to read out the slave state? Or do you have access to another EtherCAT master to try?

jacckk, 2022-04-25:
Unfortunately I don't have another EtherCAT master to try. The slave has a web interface but no obvious EtherCAT related settings, it does say it's operational when running simple-ng.

Does it sound like this could be slave hardware or software issue?

nakarlsson, 2022-10-18:
@jacckk , can we close this issue?

jacckk, 2022-10-30:
Yes. This issue is definitely caused by a failure in the slave device rather than SOEM. I tested with TwinCAT and got a similar failure. The slave was returned to the manufacturer who are investigating.

closed 2022-10-30 by jacckk