OpenEtherCATsociety/SOEM #611: Dunkermotoren BG95 Pro Ethercat Servo

kamilndbka, 2022-04-27:
Dear,
I tried to work once again on the SOEM for Turck IO and 2 Dunkermotoren servo - issue #589 
I implemented once again the init code with ROS.

Please find code shown below:
```

int GatewayNode::ethercatInit()
{
   
   /* ROS_INFO_STREAM_COND(debug_mode,"EtherCAT initialization...");

    /* initialise SOEM, bind socket to ifname */
    if (ec_init(interface_name.c_str()))
    {
	    ROS_INFO_STREAM_COND(debug_mode,"ec_init on " << interface_name << " succeeded");
        /* find and auto-config slaves */

        /** network discovery */
        if ( ec_config_init(false) > 0 ) //sprawdza czy siec wstała, czy udało sie skonfiguraowac port, sprawdzic!
        {
	        ROS_INFO_STREAM_COND(debug_mode,ec_slavecount << " slaves found and configured");usleep(100);

        }  

        for (int i=0; i<ec_slavecount; i++) 
            {
                ROS_INFO_STREAM_COND(debug_mode,"STATE: " << ec_slave[i].state);usleep(100);
            }

	
            ec_statecheck(0, EC_STATE_PRE_OP,  EC_TIMEOUTSTATE); //sprawdza stan slavów
            ec_statecheck(1, EC_STATE_PRE_OP,  EC_TIMEOUTSTATE); //sprawdza stan slavów
            ec_statecheck(2, EC_STATE_PRE_OP,  EC_TIMEOUTSTATE); //sprawdza stan slavów

            for (int i=0; i<ec_slavecount; i++) 
            {
                ROS_INFO_STREAM_COND(debug_mode,"a STATE: " << ec_slave[i].state);usleep(100);
            }
            
            ec_slave[1].PO2SOconfig=BG95ProPDOSetup;usleep(100);
            //ec_slave[3].PO2SOconfig=BG95ProPDOSetup;usleep(1000);
            //BG95ProPDOSetup(2);usleep(100);

            /** if CA disable => automapping works */
            ec_config_map(&turck.IOmap);usleep(1000);

            for (int i=0; i<ec_slavecount; i++) 
            {
                ROS_INFO_STREAM_COND(debug_mode,"b STATE: " << ec_slave[i].state);
            }
                     

            for (int i=1; i<=ec_slavecount; i++) 
            {
                ROS_INFO_STREAM_COND(debug_mode,"Slave: " << i << "\nName: " << ec_slave[i].name 
                << "\nOutput size: " << ec_slave[i].Obits << " bits\nInput size: " 
                << ec_slave[i].Ibits << " bits\nState: " << ec_slave[i].state 
                << "\nDelay: " << ec_slave[i].pdelay << "\nHas DC: " << ec_slave[i].hasdc);
            }
            //teraz urzadzenia gotowe do pracy, są w trybie initial
            ROS_INFO_STREAM_COND(debug_mode,"Slaves mapped, state to SAFE_OP.");
            /* wait for all slaves to reach SAFE_OP state */
            ROS_INFO_STREAM(" ecstate0: "<<(int)ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE));
            ROS_INFO_STREAM(" ecstate1: "<<(int)ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE));
            ROS_INFO_STREAM(" ecstate2: "<<(int)ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE));
            //ec_statecheck(1, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);
            //ec_statecheck(2, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);
            for (int i=0; i<ec_slavecount; i++) 
            {
[... 310 lines cut ...]
            ret=0;
            value8 = 0x02;
            ret +=  ec_SDOwrite(slave, 0x1c13, 0x00, false, sizeof(value8), &value8, 10*EC_TIMEOUTSAFE);usleep(300);
            ROS_INFO_STREAM(" 1c13:00, 0 ret: "<<(int)ret);

           

            ec_SDOread(slave,0x1c13, 0x00, 0, &kk, &value8,10*EC_TIMEOUTSAFE );
            ROS_INFO_STREAM(" 1c13:00 read , 1 ret: "<<(int)value8);
            ec_SDOread(slave,0x1c12, 0x00, 0, &kk, &value8,10*EC_TIMEOUTSAFE );
            ROS_INFO_STREAM(" 1c12:00 read , 1 ret: "<<(int)value8);

    return 1;
}
```

All BG95ProPDOSetup writes are corretly provided - all rets are 1. 
The problem (in my opinion) begins with ec_config_map cause there is no possibility to send data to 1c00.
Then servo can't get into OP state - last state is 6 or 22 and last do_while loop is done 200 times.
I checked 1C32:1 - mode is free run (return 0) and 1C32:5 - ret is 5 (free run and dc_mode). same for 1c33.
I also checked is PDO datas are correcttly assigned and they are (send PDO's=received)

Do You have any ideas what should be done? is there any other way to do it ?

[a.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8575560/a.zip)

ArthurKetels, 2022-04-28:
The error returned from the SDO 0x1c00:00 upload is:
`{0x06020000, "The object does not exist in the object directory"`
This object is mandatory. Therefore this is a bug in the slave firmware. Ask the vendor for a firmware upgrade.

I am not going to change SOEM to take into account every silly slave firmware bug out there. There would be no end to it. This is a very simple bug, and easy to solve for the vendor.

You could make code to circumvent this issue yourself.

```
int ecx_readPDOmap(ecx_contextt *context, uint16 Slave, uint32 *Osize, uint32 *Isize)
{
   int wkc, rdl;
   int retVal = 0;
   uint8 nSM, iSM, tSM;
   uint32 Tsize;
   uint8 SMt_bug_add;

   *Isize = 0;
   *Osize = 0;
   SMt_bug_add = 0;
   rdl = sizeof(nSM); nSM = 0;
   /* read SyncManager Communication Type object count */
   wkc = ecx_SDOread(context, Slave, ECT_SDO_SMCOMMTYPE, 0x00, FALSE, &rdl, &nSM, EC_TIMEOUTRXM);
   /* positive result from slave ? */
   if ((wkc > 0) && (nSM > 2))
   {
   ....
   }
---  new code ---
  else 
  {
     --- your alternative code ----
     if wkc == 0 read error code
     if error code == 0x06020000
    {
         do for each iSM (mostly 2 and 3)
               /* read the assign PDO */
               Tsize = ecx_readPDOassign(context, Slave, ECT_SDO_PDOASSIGN + iSM );
               /* if a mapping is found */
               if (Tsize)
               {
                  context->slavelist[Slave].SM[iSM].SMlength = htoes((uint16)((Tsize + 7) / 8));
                  if (tSM == 3)
                  {
                     /* we are doing outputs */
                     *Osize += Tsize;
                  }
                  else
                  {
                     /* we are doing inputs */
                     *Isize += Tsize;
                  }
               }
    }
  } 
```

Of course this would only work in some cases. Working around bugs from others is doomed to fail.

kamilndbka, 2022-04-28:
Thank You for Your explanations,
There is no way that Vendor will update the firmware - their opinion is that everything is correct, and they don't support other environments than Beckhoff and Twincat.

Can You explain where Your code ought to be implemented? After Po2So or just in PreOp?
How to use context with my code?

ArthurKetels, 2022-04-29:
I would leave such a vendor and look for one that is more in line with the spirit of EtherCAT. As an EtherCAT licensee they legally  bound themselves to provide products that adhere to the published standards.

For the code, I thought the suggestion above was clear. It even states the header of the function involved ecx_readPDOmap(). Look it up in the SOEM code.

nakarlsson, 2022-10-18:
@kamilndbka , can we close this issue?

nakarlsson, 2022-11-17:
Close due to inactivity

closed 2022-11-17 by nakarlsson