OpenEtherCATsociety/SOEM #613: Can not enter Operation Enable state  with no error when Enable motor

DamonGod, 2022-05-11:
Dear sir,

I met a problem that puzzled me for a long time and I couldn't find what the problem was.

When I send the Controlword 0x6/0x7/0xf successively through SOEM, the returned Stateword 0x31/0x33 shows that it enters the Switch On state but cannot enter the Operation Enable state.In the end, the Controlword becomes 0xf and the Stateword becomes 0x33.So,I cannot enable the motor.
Control_word=0x6;
State_word=0x31;
Control_word=0x7;
State_word=0x33;
Control_word=0xf;
State_word=0x33;
Control_word=0xf;
State_word=0x33;
Control_word=0xf;
State_word=0x33;
Control_word=0xf;
State_word=0x33;

[image]

1. slave's ESI file:
[Synapticon_SOMANET_Node.pdf](https://github.com/OpenEtherCATsociety/SOEM/files/8667597/Synapticon_SOMANET_Node.pdf)

2Part of My EtherCAT program
[ethercat program.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8667601/ethercat.program.zip)

3.My Equipment parameters and experimental data
uname -a
./slaveinfo -map
./simple_test

[test of ethercat.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8667613/test.of.ethercat.zip)

4.wireshark of my program
[ethercat.zip](https://github.com/OpenEtherCATsociety/SOEM/files/8667619/ethercat.zip)

Your help is very much appreciated.

best regards

ArthurKetels, 2022-05-12:
This question does not seem related to SOEM but rather the application using it in combination with a slave. You could ask other users here of there experience, there are a few that have used Synapticon with SOEM successfully. Use the search function in issues.

DamonGod, 2022-05-13:
Thank you very much for your help. I have solved the problem.

best regards

nakarlsson, 2022-05-13:
We'll close since solved

closed 2022-05-13 by nakarlsson