OpenEtherCATsociety/SOEM #615: Compiling program that uses math.h functions

krajiskiftp5, 2022-05-15:
I am getting this error when compiling my program: 

[ 86%] Built target soem
Consolidate compiler generated dependencies of target simple_test
[ 93%] Building C object test/linux/simple_test/CMakeFiles/simple_test.dir/simple_test.c.o
[100%] Linking C executable simple_test
/usr/bin/ld: CMakeFiles/simple_test.dir/simple_test.c.o: in function `find_l':
simple_test.c:(.text+0x2449): undefined reference to `sqrt'
collect2: error: ld returned 1 exit status
make[2]: *** [test/linux/simple_test/CMakeFiles/simple_test.dir/build.make:98: test/linux/simple_test/simple_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:276: test/linux/simple_test/CMakeFiles/simple_test.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

I'm trying to use sqrt and other math.h functions in my program.
I guess I should modify my Makefile but I don't really now how.
Thanks in advance.

nakarlsson, 2022-05-16:
You need to link with libmath

```
       #include <math.h>

       double sqrt(double x);
       float sqrtf(float x);
       long double sqrtl(long double x);

       Link with -lm.
```

I'm not 100% but try adding it here in CMakeFiles.txt

elseif(UNIX AND NOT APPLE)
  set(OS "linux")
  set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall -Wextra -Werror")
  set(OS_LIBS pthread rt **m**)

krajiskiftp5, 2022-05-16:
Thanks for the response.
Fixed it by adding -lm to CmakeFiles.txt

elseif(UNIX AND NOT APPLE)
  set(OS "linux")
  set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall -Wextra -Werror **-lm**")
  set(OS_LIBS pthread rt)

closed 2022-05-16 by krajiskiftp5