OpenEtherCATsociety/SOEM #616: Cannot read PDO data

krajiskiftp5, 2022-05-20:
Hello.
I am using red_test example and getting wkc of 2.

This is the stdout:

1 slaves found and configured.
Slave:1 Name:TQ870 Output size: 64bits Input size:152bits State: 4 delay:0.1
         Out:0x5560e6225280,   8 In:0x5560e6225288,  19
Calculated workcounter 3
Request operational state for all slaves
try number 1
Operational state reached for all slaves.
Processdata cycle  4896 , Wck   2, DCtime 706368740139567280, dt        17280, O: 00 00 00 00 00 00 00 00 I: 00 00 00 00 00 00 00 0

I have tried blocking LRW and here are different wireshark captures.
Slave is a hybrid stepper driver e5-hs68d:
http://m.si.hybridservos.com/uploads/20168473/PDF/EtheCat-closed-loop-stepper-drive-E5-CAT-HS68D-user-manual.pdf

[cap.tar.gz](https://github.com/OpenEtherCATsociety/SOEM/files/8740565/cap.tar.gz)

ArthurKetels, 2022-05-20:
The slave seems to behave somewhat strange. It goes to OP mode without any complaints but it is not activating the input SM. So any FMMU read to this SM will not get triggered and hence no WKC. If something in the set-up is wrong the slave should refuse to go to safe-OP, let alone OP. 

But one thing from the ESI file:
```
-<Mailbox DataLinkLayer="1">
-<CoE PdoConfig="true" PdoAssign="true" PdoUpload="false" SdoInfo="false">
-<InitCmd>
<Transition>PS</Transition>
<Index>#x6060</Index>
<SubIndex>0</SubIndex>
<Data>08</Data>
</InitCmd>
</CoE>
</Mailbox>
```
Before requesting safe-OP you should write SDO index 0x6060 subindex 0x00 with 0x08. This sets the operation mode in DS402 slaves. I do not know if this solves your problem, but it is worth trying.

krajiskiftp5, 2022-05-20:
I have tried it, and it doesn't work.

slave info, just in case:
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp0s25 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:TQ870
 Output size: 64bits
 Input size: 152bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:0.1.0.0
 Configured address: 1001
 Man: 0000ffff ID: 00000001 Rev: 00000001
 SM0 A:1000 L:  64 F:00010026 Type:1
 SM1 A:1100 L:  64 F:00010022 Type:2
 SM2 A:1200 L:   8 F:00010064 Type:3
 SM3 A:1400 L:  19 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   8 Lsb:0 Leb:7 Ps:1200 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000008 Ll:  19 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:01 Act:01
 FMMUfunc 0:3 1:1 2:2 3:0
 MBX length wr: 64 rd: 64 MBX protocols : 04
 CoE details: 0d FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6040:0x00 0x10 UNSIGNED16   Control Word
  [0x0002.0] 0x607A:0x00 0x20 INTEGER32    Profile Target Position
  [0x0006.0] 0x60B8:0x00 0x10 UNSIGNED16   Touch Probe Function
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0008.0] 0x603F:0x00 0x10 UNSIGNED16   Last Error Code
  [0x000A.0] 0x6041:0x00 0x10 UNSIGNED16   Status Word
  [0x000C.0] 0x6061:0x00 0x08 INTEGER8     Modes of Operation Display
  [0x000D.0] 0x6064:0x00 0x20 INTEGER32    Actual Position
  [0x0011.0] 0x60B9:0x00 0x10 UNSIGNED16   Touch Probe Status
  [0x0013.0] 0x60BA:0x00 0x20 INTEGER32    Touch Probe 1 Positive Value
  [0x0017.0] 0x60FD:0x00 0x20 UNSIGNED32   Digital Inputs
End slaveinfo, close socket
End program

krajiskiftp5, 2022-05-21:
FMMUfunc 0:3 1:1 2:2 3:0, is a somewhat uncommon order right?
I've mainly seen 0:1 1:2 2:3 3:0
Maybe a firmware bug?

ArthurKetels, 2022-05-21:
It is indeed not common to have this FMMU set-up, but in usual cases the slave can not control the use of FMMU's. In other words, the master can use slave FMMU's any way it likes. The slave only has control over the SM's. You could try to read and print out the SM control and status registers after transition to OP. Then at least we can see if the slave disabled the input SM.

An other reason could be that the slave requires SYNC0 set-up. From the ESI file it looks like it supports both free-run and DC SYNC0. But I have seen bugs like this before.

A test run with TwinCAT and then capturing the WireShark trace can reveal the correct way to start this slave up.

krajiskiftp5, 2022-05-21:
I will try using TwinCAT and I'll post results, in the mean time, how do I setup SYNC0? 
I've seen it mentioned in the users manual linked above but they didn't go into specifics and I have failed trying to set it up myself. Could you write which line I need to add in my code?

nakarlsson, 2022-10-18:
@krajiskiftp5 , any update about TwinCAT? Or can we close this issue.

krajiskiftp5, 2022-10-29:
Issue was most likely caused by bad firmware, stopped experimenting because onboard dc-dc power chip burned out.

closed 2022-10-29 by krajiskiftp5